MrChromebox/firmware #2: Hibernation is Broken.

ReddestDream, 2016-09-02:
Hibernation on all HSW/BDW Chromebooks appears to be broken in all OSes. Following Hibernation, Battery drains excessively and the system never fully shuts down. This appears to be related to the fact that the S4 power state is not really supported, only S3 and S5. Perhaps S4 could be fixed as Coreboot reports that it is supported. Otherwise, perhaps S4 should be disabled in Coreboot to avoid confusion. Thanks!

MattDevo, 2016-09-02:
I'd characterize this by saying that hibernation works (since the device fully shuts down and restores to the previous state correctly), but not all components are being put into the correct low-power state, leading to excessive power drain (basically equal to S3?)

ReddestDream, 2016-09-03:
Yeah. If you would like, I'll edit the title accordingly.

ReddestDream, 2016-09-20:
Recent fixes have at least corrected USB-related drain. Further testing is needed to see how much improvement that brought.

ReddestDream, 2016-09-22:
Recent fixes have brought at least 50% improvement. Closing this issue for now. Might still get minor fixes, but the major issue is resolved.

closed 2016-09-22 by ReddestDream