MrChromebox/firmware #3: Add UEFI Shell and Additional Drivers to Duet.

ReddestDream, 2016-09-02:
Duet is currently lacking a built-in UEFI shell and is missing filesystem drivers other than FAT32, NTFS, and EXT4 (most notably, HFS+ is missing). A UEFI shell would give users another way to get more information about their systems and would also provide a means of accessing filesystems if the system becomes unbootable (part of why including more drivers is important). Additionally, it would provide another method of finding and starting EFI binaries, especially bootloaders that are not automatically detected. Thanks!

MattDevo, 2016-12-08:
no point in doing this since moving away from Duet / towards CorebootPayloadPkg

closed 2016-12-08 by MattDevo

ReddestDream, 2016-12-08:
Part of CorebootPayloadPkg work, yeah.