MrChromebox/firmware #4: Fix for Booting Linux via UEFI.

ReddestDream, 2016-09-02:
As it stands currently, without NVRAM support, systems running these firmwares are unable to boot Linux after running an installer from a standard Linux distro. This is because these distros do not install EFI GRUB to /EFI/BOOT/BOOTX64.EFI, the only place this firmware looks by default for a bootloader. Linux does this to avoid issues with Windows potentially overwriting that location, but it means that manually fixes are required to get Linux booting on these firmwares. :/

There seem to be two possible approaches other than implementing some form of NVRAM: 1. Add additional path finding to Duet so that it can find Linux (and other) bootloaders. 2. Provide a Boot Management Utility, such as rEFInd in the firmware so that all installed Bootloaders will be automatically detected and displayed in simple GUI for users. Thanks!

palob, 2017-02-23:
(In case you aren't dualbooting) a much simpler and less intrusive fix is to just move the GRUB file you want to have booted to ``/EFI/BOOT/BOOTX64.EFI``.

MattDevo, 2017-02-23:
@palob yes, which is why it's listed as the preferred workaround on the FAQ  :)

MrChromebox, 2019-01-05:
is this still an issue?

closed 2019-01-05 by MrChromebox

ibedani, 2019-12-20:
mr chromebox firmware can't boot windows 10 uefi  automatically unless efi is set manually. pls help with steps. am new here

ibedani, 2019-12-20:
this the video link
https://youtu.be/WD_w70HYgKU