MrChromebox/firmware #5: HDMI Audio Does not work properly on Windows or OS X.

ReddestDream, 2016-09-03:
HDMI Audio is broken on HSW/BDW on both Windows and OS X. On Windows, CS has found a workaround using a third party program to get the EDID data and then use that to start HDMI audio. This has to be done on a per-display basis.

https://www.reddit.com/r/chrultrabook/comments/4j42ln/tutorial_enable_hdmi_audio_in_windows_no_linux/

Thanks!

coolstar, 2016-11-16:
Fixed via IGD OpRegion for Ivy Bridge and Haswell.

Broadwell has a different "chipmunk" issue.

MattDevo, 2016-11-16:
edited title to reflect working state of HDMI audio under Windows as of 11/16 UEFI firmware release

ReddestDream, 2016-11-16:
I'll test soon on OS X to see if it has been fixed as well.

ReddestDream, 2016-11-17:
Confirmed. Still broken on OS X.

coolstar, 2017-01-24:
This is a non-issue in the firmware. HDMI audio in macOS doesn't work on my non-chromebook Lenovo Flex 3 hackintosh either (Intel HD 520 with Phoenix UEFI firmware). macOS just needs extra configuration and possibly patches.

ReddestDream, 2017-01-24:
If we can find a way to make it work w/o any changes to the firmware, and there's nothing to be gained by adding anything to the firmware (too OS X specific, as with the graphics, and makes more sense in an SSDT), then I will gladly close this issue.

xjbhenry, 2020-05-18:
@ReddestDream I recently figured out the reason. Basically pci device `8086:160c` was not named as `HDAU` in DSDT. So I named that device as HDAU and HDMI audio works now.
You can use this dsdt with AppleALC.
https://github.com/xjbhenry/asus-chromebox-cn62-hackintosh/blob/master/ACPI/dsl/SSDT-HDAU.dsl

ReddestDream, 2020-05-18:
@xjbhenry The fix was not that simple originally. Tho it is good to know that maybe AppleALC fixes this problem now . . .

xjbhenry, 2020-05-18:
@coolstar I think we need to add `HDAU` in SSDT.

xjbhenry, 2020-05-18:
> @xjbhenry The fix was not that simple originally. Tho it is good to know that maybe AppleALC fixes this problem now . . .

Yes, AppleALC can add `hda-gfx` and `layout-id` for you, and make these values consistent between `HDEF` and `HDAU`. So you don't need to manually set them.

coolstar, 2020-05-18:
Oh lol I forgot about this issue, but yes I can confirm.

I’ve been using OpenCore to add HDAU via SSDT for an eternity and had HDMI audio working in macOS, but I forgot about it lol

ReddestDream, 2020-05-18:
Guess we can close this then?

coolstar, 2020-05-18:
Well might be a good idea to add HDAU to Coreboot's ACPI tables

ReddestDream, 2020-05-18:
Ah. Maybe . . .

MrChromebox, 2020-05-18:
> Well might be a good idea to add HDAU to Coreboot's ACPI tables

for what device(s)? There's maybe 1-2 that can run MacOS passably? and if one can work around it in OpenCore, same as one would on a normal PC, then I'm not sure how much of an issue this is

coolstar, 2020-05-18:
Coreboot could have the HDAU for HSW/BDW devices with the ADR set. Then the
opencore SSDT only needs to add the DSM rather than the entire device (or
clover can add the macOS properties without SSDT with HDAU added)

xjbhenry, 2020-05-18:
For Asus Chromebox 2 (CN62) GUADO, I think potentially we could add HDAU for all Haswell/Broadwell

MrChromebox, 2020-05-18:
> Coreboot could have the HDAU for HSW/BDW devices with the ADR set. 

sure, I can do that. Just need the snippet and where to add it

coolstar, 2020-05-18:
Something like this could work

```
Device (HDAU)
        {
            Name (_ADR, 0x00030000)  // _ADR: Address
        }
```

Could probably add it to src/northbridge/intel/haswell/acpi/haswell.asl and src/soc/intel/broadwell/acpi/systemagent.asl underneath the include for drivers/intel/gma/acpi/pch.asl

(Don't stick it in drivers/intel/gma/acpi/ though because Skylake+ does not have a separate HDAU device. They run HDMI audio through the normal HDEF as a 2nd codec)

MrChromebox, 2020-05-19:
done

MrChromebox, 2020-06-07:
confirmed working in 4.12 release; closing

closed 2020-06-07 by MrChromebox