MrChromebox/firmware #7: Video Playback Failure on BDW in OS X After Sleep.

ReddestDream, 2016-09-03:
After sleep with graphics acceleration working, video playback breaks on OS X, resulting in hangs. QE/CI appears to continue to function normally otherwise. This seems to affect all BDW Chromebooks once graphics acceleration is working regardless of igd configuration. As of yet no precise cause has been determined. At this point, a firmware issue appears highly plausible.

Connected to this may be the need to sleep/wake some Broadwell Chromebooks in order to get video out on OS X when graphics acceleration is enabled. This does not affect the C740, however, limiting my ability to test this issue. The video playback issue reported here does, however, also affect the C740.

ReddestDream, 2016-11-17:
This issue still exists on the 11/17 firmware.

MrChromebox, 2025-12-03:
Closing due to time / lack of follow-up

closed 2025-12-03 by MrChromebox