MrChromebox/firmware #10: Duet Settings Do Not Save.

ReddestDream, 2016-09-05:
The Settings in Duet do not save. :/

This could be fixed via NVRAM or perhaps, as proposed by CoolStar, but saving them in the CMOS.

palob, 2017-02-23:
This is annoyance level 10^28.

dalvarezsmiet, 2018-02-14:
Is there any progress on this issue? It's amazing how a pain in the neck it can become...

lis6502, 2018-07-01:
Issue still exists on Lenovo x131e (chromebook Stout), settings for EFI are not saved desptie of 'commiting' changes.

lis6502, 2018-07-01:
@MrChromebox i've seen that there's Closed next to some replies in this issue, so i though it's been... closed :)
I am not hardware hacker as you, but there's upstream versions of Coreboot( not specifically in Stout version)- did you managed to check how they solved saving settings?
startup.nsh resides somewhere though- maybye it's an option?
I would not mind to workaround with saving settings directly to some /EFI/config file on selected medium and querying for this file's values when booting up- if NVram is hard to access.

MrChromebox, 2018-12-26:
not going to close yet, but will be resolved in next release :)

MrChromebox, 2018-12-31:
(preliminary) NVRAM support added in 2018.12.30 release

closed 2018-12-31 by MrChromebox

moriel5, 2019-09-16:
From my tests, settings do not save on any versions since the preliminary NVRAM support has been added, at least on `gnawty`.

MrChromebox, 2019-09-16:
I just tested the 8/22 release on SWANKY, and no issues with NVRAM saving. Try clearing your NVRAM settings using the script option and see if that helps

moriel5, 2019-09-16:
Thanks, I'll try that..

moriel5, 2019-09-16:
Nope, changes are still not being saved.
I had tested by trying to change the console size, could there be an unrelated bug with it?

MrChromebox, 2019-09-16:
the bootorder is the only thing hooked up; timeout, console size, language etc are not saved

moriel5, 2019-09-16:
Ah, are there any plans for saving those as well?
Since I'd like to help translate Coreboot to hebrew.

MrChromebox, 2019-09-16:
as I mentioned in the GUI issue, the current Tianocore package being used is now deprecated, so I don't plan on doing any work to it until switching over to the newer UefiPayloadPkg. 

I'm not sure what the language support is like currently, but that's Tianocore, not coreboot.

moriel5, 2019-09-16:
Sorry for not being clear.
I had meant that that I'd like to do so upstream (which probably means in this case, `UefiPayloadPkg`), since that should also remove the burden from your shoulders.

Venji10, 2021-01-31:
Hello, I just installed coreboot with tianocore on my Thinkpad X220. It works perfectly fine but I have to select the grub efi file to boot manually everytime since I neither can save the boot order and entries in tianocore nor with efibootmgr from linux.

Do I need to enable some extra config for NVRAM support or is this not the issue I have?

MrChromebox, 2021-01-31:
> Do I need to enable some extra config for NVRAM support or is this not the issue I have?

you need `CONFIG_SMMSTORE=y` (and not SMMSTORE_V2)

Venji10, 2021-01-31:
Thank you, that fixed it!