MrChromebox/firmware #14: Fix Opregion To Eliminate Need for VBIOS.

ReddestDream, 2016-09-20:
If Opregion could be fixed, all need for the VBIOS would eliminated. Currently, the VBIOS is only needed by the Intel Driver on Windows. If Opregion could be properly implemented, there would no longer be a need for the VBIOS. This might allow us to move back to Coreboot TianoCore Payload instead of Duet since there would be no reason to keep SeaBIOS around to run the VBIOS.

MattDevo, 2016-09-20:
SeaBIOS isn't needed to run the VBIOS, coreboot can do that just as well; SeaBIOS is only needed to load Duet, since the native Tianocore coreboot payload (CorebootPayloadPkg) is far less functional at this point

coolstar, 2016-09-22:
@MattDevo yes, however, when I ran some tests that doesn't work. When I ran the VBIOS in coreboot and had corebootpayloadpkg, the intel driver in Windows still had problems. Something was being erased from memory between coreboot/corebootpayloadpkg

MattDevo, 2016-09-22:
@coolstar perhaps a difference in where it gets mapped into memory?

coolstar, 2016-11-16:
Done. OpRegion is implemented and working on Sandy Bridge, Ivy Bridge, Haswell and Broadwell platforms :)

Goodbye Video BIOS!

MattDevo, 2016-11-16:
will be implemented in 11/16 firmware release :)

closed 2016-11-16 by MattDevo