MrChromebox/firmware #15: Lid open in S4/S5 = momentary power on, then off

Paviluf, 2016-09-23:
As the title says when you open the lid in S4/S5 the chromebook momentary power on, then off and you need to press the power button to power it on. Seems to happen only for Haswell Chromebooks.

ReddestDream, 2016-09-23:
Can confirm this issue on C720 (also HSW). Works properly on C740 (BDW).

ReddestDream, 2016-09-23:
The only way to resolve the issue (temporarily) is to close the lid during power on (before OS loads). This will fix the issue on HSW until hibernation is initiated with the lid open again. The issue seems to be triggered by hibernation occurring when the lid is open.

Paviluf, 2016-10-23:
Maybe I'm totally wrong but maybe this is somewhat related to this https://github.com/MattDevo/firmware/issues/18 because it look like to be the same behaviour. I explain :

**With Full ROM**
- Plug AC Power => Start (sometimes)
- Open Lid in S4/S5 => Momentary power on, then off

**With Stock + Legacy RW**
- Plug AC Power => Momentary power on, then off (sometimes)
- Open Lid in S4/S5 => Start

It seem like there is some kind of inversion in the handling of the lid and the power ac plug in for some chromebooks.

ReddestDream, 2016-10-23:
So, as interesting as that may seem, I am able to get "Start" on "Open Lid in S4/S5" on the Full ROM without AC via the fix I noted [here](https://github.com/MattDevo/firmware/issues/15#issuecomment-249197175). I have not done extensive testing of this issue on Stock + Legacy RW.

Paviluf, 2016-10-23:
Maybe it's just for the Falco ? 

>  I have not done extensive testing of this issue on Stock + Legacy RW.

There is no problem with the Stock + Legacy RW behaviour :) I don't know if it's intended that it momentary power on, then off when you plug AC power in S4/S5 but it stay off, that's what matter. It's just the "ON" led that "blink" for a second. maybe it's intended or maybe it's a motherboard "bug" that is workaround by the EC firmware but the thing is that it stay off. I don't think one should spend time on this because that work :)

ReddestDream, 2016-10-23:
>Maybe it's just for the Falco ?

Possibly, but all the HSW Chromebook have some issue . . .

Paviluf, 2016-12-16:
Have you some news about fixing this bug ? Is there something I can do to help fixing it ?
Thanks !

ReddestDream, 2016-12-16:
No news ATM. I plan to do more EC firmware tests when I have time.

Paviluf, 2016-12-16:
Ok thank you. If you want me to test some things not to risky I will be happy to do so :)

natloh, 2017-01-02:
Just dropping in to say I have an Acer CB3-111 gnawty that is experiencing the exact same issue. However it doesn't matter if the power adapter is connected or not. When I open the lid, the activity light turns on then off, and then the device boots.

MattDevo, 2017-01-02:
@natloh that's not the same at all -- the issue described here is that opening the lid causes the device to start to boot, then power back off.  What you're describing is normal operation - open the lid, the device turns on

natloh, 2017-01-03:
@MattDevo I should have specified that my case occurs when the device is powered on. I suppose you are right my issue is slightly different. 
While booted and I close the lid, the activity light flashes orange, implying it is in sleep mode, but when I open the lid, mine appears to be resuming from sleep then instead powers off the device and then reboots.

MattDevo, 2017-01-03:
@natloh reboot on resume from suspend is a known issue with the Full ROM firmware on Baytrail devices currently

natloh, 2017-01-03:
@MattDevo Got it. Thank you for the update! I'll keep an eye out on the updates awaiting a fix. Is there a issue that I can keep an eye on for that?

MattDevo, 2017-01-03:
@natloh there's no open issue for it right now, feel free to open one yourself :)

natloh, 2017-01-05:
The issue I was talking about has been opened here #38

Paviluf, 2017-02-09:
Hello @MattDevo ,

I asked for help on Chromium Os dev Google Group and I had an answer.
I think it will be really great if you join us.

https://groups.google.com/a/chromium.org/forum/#!topic/chromium-os-dev/NiaVZloW9Jw

Thanks !

ReddestDream, 2017-02-15:
I believe I may solved this issue, but I am seeing additional confirmation, @Paviluf.

Please see the file I sent you on Discord.

MattDevo, 2017-02-21:
fixed via EC firmware update in 2017.02.21 release

closed 2017-02-21 by MattDevo