MrChromebox/firmware #16: UEFI mis-catagorization on 9/22/16 rom

xrayk, 2016-09-23:
My Acer C910 (auron-yuna) was classified as an Acer C7xx (auron-maine?) by the UEFI rom bios installation.
The bios would come up on screen and navigation was possible, but boot would not activate.  I was locked in the bios.

Used script to return to Legacy boot.

ROM and bios working beautifully.  Thank you

MattDevo, 2016-09-23:
had you previously installed coolstar's older UEFI firmware?  I think at one point he used a single ROM for both Auron devices, which would have lead the script to misidentify your model.  Even still, there's no reason it shouldn't have booted if you had a UEFI OS installed

xrayk, 2016-09-23:
No. Only the Lewis Legacy mod. Tried again using the Kodi setup script.
Got the same misclassification on the bios menu

MattDevo, 2016-09-23:
ok, so what firmware are you running now?  If you use my script to go back
to stock and didn't use your own backup, then it likely flashed the stock
auron_paine firmware

xrayk, 2016-09-23:
I flashed your legacy 9/22 firmware.  Yes, it is probable auron_paine
firmware.  But in legacy mode, it seems to be working perfectly.

MattDevo, 2016-09-23:
both Auron firmware will work for both devices, the only differences are the backlight panel timing, and the built-in audio configuration.  If you have the wrong Auron firmware flashed, then your audio setup is likely not fully functional.

the best way to resolve this will be to flash the correct stock firmware, then reflash the correct Legacy/UEFI firmware.  Hit me up on Hangouts - mrchromebox at gmail - and I'll walk you thru it

closed 2016-09-23 by MattDevo

xrayk, 2016-09-23:
With the current setup, system and audio are perfect.  Even my fussy
bluetooth headphones are working nicely.

When it is timely, how orwhere do I find the auron-yuna firmware.   I had
used your script before.

xrayk, 2016-09-23:
Note that I am locked at 100% screen brightness.  Further, cannot use your
script to return to stock auron-yuna firmware.
Could I change the hardware ID?