MrChromebox/firmware #18: Chromebook boot sometimes when you plug the power adapter when the Chromebook is completely off

Paviluf, 2016-09-29:
I thought it was fixed but, unfortunately, it's not. The Chromebook still boot sometimes when you plug the power adapter when the Chromebook is completely off.

ReddestDream, 2016-09-30:
I've seen this happen on C740 after the battery is unplugged. Seems to be normal if that's all it is for you.

Paviluf, 2016-09-30:
It never happen with stock firmware + Legacy RW. Note that with stock firmware + Legacy RW sometimes when you plug the power adapter when the Chromebook is completely off, the power led switch on (like the Chromebook is starting) but switch off after a second.

Paviluf, 2016-10-23:
Could it be related to this that is maybe not working with the full rom ?

https://chromium.googlesource.com/chromiumos/platform/ec/+/0596b551dfd45dd4af2e8f7a0baad1141d25bfb8

ReddestDream, 2016-10-23:
Well, there does seem to be an EC issue on HSW. The EC firmware, the source to which you link, is unmodified, however. Custom firmwares are using the RO EC firmware that was already present. It is not altered when the main firmware is flashed. This is something we are looking into (also to fix the keyboard so that it doesn't require a special driver for Windows/OS X), but currently, it is the same as the stock RO firmware.

Paviluf, 2016-10-23:
It will be very great if with can update the EC firmware but that's don't what I meant. For this problem, since with Legacy RW and Full Rom the EC firmware is the same and the Legacy RW work perfectly it can not be a EC firmware problem. I meant that, maybe, the Full Rom don't interpret correctly the EC firmware commands for this case. Maybe there is something "swaped" that explain this behaviour ?

**With Full ROM**
- Plug AC Power in S4/S5 => Start (sometimes) https://github.com/MattDevo/firmware/issues/18
- Open Lid in S4/S5 => Momentary power on, then off https://github.com/MattDevo/firmware/issues/15

**With Stock + Legacy RW**
- Plug AC Power in S4/S5 => Momentary power on, then off (sometimes)
- Open Lid in S4/S5 => Start

>  I have not done extensive testing of this issue on Stock + Legacy RW. (https://github.com/MattDevo/firmware/issues/15#issuecomment-255596785)
By the way, as I say there, there is no problem with the Stock + Legacy RW behaviour ;)

Paviluf, 2016-10-23:
Let's imagine that there is something different (bug ?) in the EC firmware of the Falco that make it send the wrong command when you plug the AC power or open the lid that is "fixed" by the stock firmware. That could explain this behaviour maybe ?

ReddestDream, 2016-10-23:
Probably, yeah. All the HSW books have some issue . . . :/

Like I said, something we are looking into . . .

Paviluf, 2016-10-23:
> Like I said, something we are looking into . . .

Ok I will wait, it's just some paper cut ;)

Are you sharing the infos to compile and flash the EC firmware somewhere ? I want to make some changes to the fan handling.

ReddestDream, 2016-10-23:
>Are you sharing the infos to compile and flash the EC firmware somewhere ? I want to make some changes to the fan handling.

I don't know how to change it. Are you familiar with ectool? If not, it's something you should look into first . . .

Paviluf, 2016-10-23:
I meant just sharing the infos to compile and flash the EC firmware. I already know (I think) what to change for the [fan handling](https://chromium.googlesource.com/chromiumos/platform/ec/+/firmware-falco_peppy-4389.B/common/thermal.c). For now I use a systemd timer that trigger every 5s a little script to control the fan with ectool according to the temps. That work pretty well, my Falco is always loud by default and very very quiet with my script, but a proper handling directly in the firmware will be better.

ReddestDream, 2016-10-23:
Okay. First, you'll need to set up the Chromium chroot to compile EC firmwares. You'll want to find the latest EC source that was used to make a firmware for your board. That's probably this one:

https://chromium.googlesource.com/chromiumos/platform/ec/+/firmware-falco_peppy-4389.81.B

Once you make a firmware, you'll need to use flashrom to flash the EC firmware and only the RW portion. This requires very specific options. Then you reboot the EC into the RW firmware to load it. You don't touch RO EC firmware since Custom Firmwares don't reboot the EC from RO to RW like Stock firmware does. This way, you can still boot if your EC firmware is screwy without needing a servo. I can get you all the commands for this process once you are able to make a firmware and find them all again . . .

Just be aware that setting the Chromium chroot is a very messy, intensive process and that it is very possible to flash the EC firmware improperly. The only way to recover should you mess up this process is with a Debug Servo. The chip that holds the EC firmware is not clippable to my knowledge.

coolstar, 2016-10-25:
The stock behavior is due to how depthcharge is implemented. On boot, depthcharge checks the lid state and switches off the system if it detects the lid is shut. 

The way our firmwares behave now are more similar to the way a normal UEFI is set up. If we have our firmwares switch off the system while the lid is shut, it will break headless operation for those who use it (have the chromebook running with the lid shut and initiate a remote reboot).

The way depthcharge is implemented is completely non-standard, so we should not implement it the way stock does it.

Paviluf, 2016-10-27:
@ReddestDream Sorry for the late answer, I was away. Thank you but it seem very complicated and like you said don't worth it just for the fan. Anyway I have my scripts that work perfectly. I just thought that was easy and safe when you said this

> I have confirmed that we can recompile and reload the EC firmware safely w/o bricking

@coolstar Thanks for the explanation. The real question is why plug in the ac power trigger a boot sometime ? Do you know if it's it a EC firmware feature or a bug or something else ?

MattDevo, 2016-11-08:
@Paviluf can you provide exact steps for how to reproduce this?  Trying to do so with a Peppy but not having any luck

Paviluf, 2016-11-08:
I just use the chromebook normally, power it off, close the lid, plug ac power and sometimes a boot is triggered. The thing is that appear to happen randomly. I even thought it was gone when you make me test a firmware but that re-happen a few days later. Maybe it happen more frequently when the battery is low, I don't know. I can make more tests if you want.

MattDevo, 2016-11-08:
@Paviluf if the battery has fully drained, that is the expected behavior, since the EC will have lost power and will power on when it is restored via AC power

Paviluf, 2016-11-08:
I never let it fully drained. I meant 10/15% when I said low battery.

ReddestDream, 2016-11-09:
Yeah. IDK. It might be an older RO EC firmware. I have not ever had it happen unless the battery is dead. I can't reproduce the turning-on-when-AC-is-plugged-in thing.

@Paviluf Glad to hear it is sporadic/random. Kinda explains why I can't reproduce it . . .

Paviluf, 2016-11-09:
I will try to find a pattern to reproduce the problem. I hope that ec firmware flash will be available soon :)

ReddestDream, 2016-11-11:
@Paviluf I have managed to reproduce this issue. You aren't crazy! :)

It happens randomly when plug in with the lid shut. The battery must be somewhat discharged, but it need not be totally discharged.

Paviluf, 2016-11-11:
> You aren't crazy! :)

Yeah I knew that :D

Paviluf, 2016-12-16:
Have you some news about fixing this bug or providing ec firmware flash ? Is there something I can do to help fixing it ?
Thanks !

ReddestDream, 2016-12-16:
No news ATM. I plan to do more EC firmware tests when I have time.

Paviluf, 2016-12-16:
Ok thank you. If you want me to test some things not to risky I will be happy to do so :)

ReddestDream, 2017-01-24:
I am still looking into this. I've determined from the EC console that when a HSW book fails to power on on lid open, there is a switch from G3-S5 but the process of wakeup seems to end prematurely and the system ends up back in G3. On BDW and BYT, the process completes correctly and the device turns on fully (ends in S0). I'm still not sure what accounts for this difference in behavior. Back porting of several power and button related fixes (since the lid uses a power button event) from BDW has still not brought any improvement tho at least I've ruled more out. :/

I figured out how to fix the keyboard tho. An update using a Software Sync-like system to update the EC firmware on HSW/BDW/BYT devices is coming soon to bring the keyboard fix. :)

Paviluf, 2017-01-25:
Thank you very much for your amazing work. It's awesome that will be able to update the EC firmware !

Paviluf, 2017-02-10:
> Checking closer, you're probably waking from hibernate -- unless you modified EC FW, falco EC will go to hibernate after 1 hour of off time w/ AC unplugged. A wake event (typically AC attach, lid open, power button press -- but I'm not sure exactly what we implemented on falco, it's so old) is then handled very similar to EC cold boot. Again, if you want to override this behavior, you'll probably want to pretend that SYSTEM_RESET_LEAVE_AP_OFF is set. 
https://groups.google.com/a/chromium.org/forum/#!topic/chromium-os-dev/NiaVZloW9Jw

Seem to be a feature finally ! It happen only after an hour so it's seem to be the intended behaviour.

ReddestDream, 2017-02-10:
Glad you could confirm that. Are you ready to close this issue? ;P

Paviluf, 2017-02-11:
yes ;)

closed 2017-02-11 by Paviluf