MrChromebox/firmware #20: bootmode questions

DennisLfromGA, 2016-10-15:
I have an Acer C710 (parrot) which you now support, in part at least I think.

Just a couple of questions to clarify what I'm interpreting on the [#bootmodes](https://mrchromebox.tech/#bootmodes) page on Mr.Chromebox.

Under: **SeaBIOS: The Legacy Boot Payload**
> Note: If only one boot device is available -- either no USB/SD devices are attached, or they are not properly detected/initialized by SeaBIOS -- then the boot menu prompt ("Press ESC for boot menu") will not be shown, and the internal storage will be booted. *__In the case where ChromeOS is still installed/is the only OS installed, this can result in the screen showing text indicating a kernel panic, or simply a blank screen. If this happens, simply reboot and try using a different USB/SD device to install/boot from.__*

Does this mean that you can no longer boot ChromeOS at all on the internal disk, only external OSes?

Under: **UEFI and the Future**

> As of 2016-09-06, the Firmware Utility Script now offers Full ROMs offering either Legacy Boot or UEFI Boot for all Haswell and Broadwell devices. UEFI (Unified Extensible Firmware Interface) is the successor to Legacy BIOS and is the firmware type used by virtually all Windows PC's (and Macs) shipped in the last 5+ years.

Can I still boot ChromeOS on the internal drive if I install this BIOS?

-----
I would love to have the ability to run other OSes, both externally and in a dual-boot scenario but I don't want to do so at the expense of running ChromeOS natively.

Thanx for all of your efforts, this is *cool* stuff,
-**DennisL**

MattDevo, 2016-10-15:
@DennisLfromGA with the Full ROM firmware (either Legacy or UEFI), you can't run ChromeOS anymore unfortunately, but you can run the open-source versions (ChromiumOS, CloudReady, etc).  Unfortunately, the C710 doesn't offer the option to boot in Legacy Mode from the stock firmware, so it's an either-or proposition.  We hope to have the ability to run ChromeOS (with auto-updates functional) via the UEFI firmware, but it's not there yet

DennisLfromGA, 2016-10-15:
K, thanx for the quick answer @MattDevo.

I've still got a working ChrUbuntu loaded on parts 6&7 and can use the old 'change kernel prorities' switching method so I'm good.

Glad I asked though since I didn't fully understand what those BIOSes/firmwares did.

I've got a Dell Chromebook 13 (i5/8GB/256GB) 'lulu' that I can use to do just about everything I want. Soon it'll have Android apps too. :)

On my 'lulu', I currently have a tweaked (john.lewis) SeaBIOS but I saw in your script that you 'Update the RW_LEGACY Firmware' and it may offer *some improvements* over the john.lewis firmware so I'm contemplating installing it. I think that can be done without disabling the write-protect.

*Thanx again for your reply,*
-**DennisL**

MattDevo, 2016-10-16:
>  I saw in your script that you 'Update the RW_LEGACY Firmware' and it may offer some improvements over the john.lewis firmware so I'm contemplating installing it. I think that can be done without disabling the write-protect. 

it does, and correct.  The FW script documentation clearly states for each function if WP needs to be disabled or not :)

closed 2016-10-16 by MattDevo