MrChromebox/firmware #21: Enable/Support TPM.

ReddestDream, 2016-10-24:
Currently, the TPM is disabled in the firmwares. It would be great if we could get to the point of enabling it. AFAIK, it is on the roadmap for around the time when NVRAM is added, but there have been several requests about it. Thanks!

coolstar, 2016-10-25:
TPM causes problems with sleep/wake, hence we have disabled it for now. Also support requires taking ownership of the TPM from the UEFI, so we would need to add support for in edk2.

ReddestDream, 2017-01-14:
TPM being enabled seems be okay on BYT, but, until the resume issue noted [here](https://github.com/MattDevo/firmware/issues/38) is resolved, nothing involving sleep/wake can be confirmed on BYT.

ReddestDream, 2017-01-14:
@coolstar The sleep/wake issue seems to have resolved with the move to CPP! \^o^/

Tests on Peppy in both Windows and Linux show no sleep/wake issues with the TPM as enabled as I can get it . . .

WRMilling, 2017-02-15:
I have found myself wanting to encrypt and I would prefer to use a TPM over a flash drive. What is the level of effort to actually get this working in the firmware?

ReddestDream, 2017-02-15:
It's on the list to investigate after the next major update.

ReddestDream, 2017-03-08:
So, TPM will not work for Bitlocker with testsigning on, which puts a bit of a damper on things . . . :/

Also, having trouble getting it inited properly by Coreboot/SeaBIOS. DepthCharge/Vboot must be doing something . . .

htruong, 2018-05-04:
>So, TPM will not work for Bitlocker with testsigning on, which puts a bit of a damper on things . . . :/

Many users of yours are using Linux anyway, so even when it doesn't work on Windows, having it working on Linux is still a boon for many :)

CiriousJoker, 2019-11-26:
My chromebook (Acer Spin 13) has these hibernate/sleep issues. Basically, sleeping/hibernation causes the fan to spin at 100% if plugged in or to shut down when on battery.
Will this be fixed at any point? Is this fixable at all?

MrChromebox, 2019-11-26:
@CiriousJoker please open a new issue since this has nothing to do with the TPM

OsirianOne, 2021-11-22:
My Acer Chromebook 15 CB3-532 will recognize the TPM as TPM 1.2.  I am curious if this has been fixed or if this issue is still open.  Seeing how its been 2 years ans 6 months. Maybe it was forgotten?  

Updated my BIOS today using MrChromebox update version is: **MrChromebox-4.14**  Full export attached:
[dmicode.txt](https://github.com/MrChromebox/firmware/files/7578292/dmicode.txt)

TPM Error output regarding the inability to initialize:
sudo cat /var/log/kern.log | grep tpm
Nov 21 04:42:41 Osiris-Banon kernel: [    0.969532] tpm_tis 00:06: 1.2 TPM (device-id 0xB, rev-id 16)
Nov 21 05:01:44 Osiris-Banon kernel: [    0.971717] tpm_tis 00:06: 1.2 TPM (device-id 0xB, rev-id 16)
Nov 21 16:23:12 Osiris-Banon kernel: [    0.956064] tpm_tis 00:06: 1.2 TPM (device-id 0xB, rev-id 16)
Nov 21 17:56:46 Osiris-Banon kernel: [    0.992458] tpm_tis 00:06: 1.2 TPM (device-id 0xB, rev-id 16)
Nov 21 22:28:41 Osiris-Banon kernel: [    0.971056] tpm_tis 00:06: 1.2 TPM (device-id 0xB, rev-id 16)

Then checking systemd (which appears to support the note from your comments in the thread that the ownership may not have been taken over so, could that be the reason it could not initialize?):
fwupd.service - Firmware update daemon
     Loaded: loaded (/lib/systemd/system/fwupd.service; static; vendor preset: enabled)
     Active: inactive (dead)
       Docs: https://fwupd.org/

Nov 21 23:27:55 Osiris-Banon systemd[1]: Starting Firmware update daemon...
Nov 21 23:27:55 Osiris-Banon fwupd[3826]: 04:27:55:0606 FuPluginUefiCapsule  SMBIOS BIOS Characteristics Extension Byte 2 is invalid -- UEFI Specifi>
Nov 21 23:27:55 Osiris-Banon fwupd[3826]: ERROR:sys:src/tss2-sys/api/Tss2_Sys_Execute.c:114:Tss2_Sys_ExecuteFinish() Unsupported device. The device >
Nov 21 23:27:55 Osiris-Banon fwupd[3826]: ERROR:esys:src/tss2-esys/api/Esys_Startup.c:216:Esys_Startup_Finish() Received a non-TPM Error
Nov 21 23:27:55 Osiris-Banon fwupd[3826]: ERROR:esys:src/tss2-esys/api/Esys_Startup.c:78:Esys_Startup() Esys Finish ErrorCode (0x00080001)
Nov 21 23:27:55 Osiris-Banon fwupd[3826]: 04:27:55:0823 FuEngine             failed to add device /sys/devices/pnp0/00:06/tpm/tpm0: failed to initia>
Nov 21 23:27:56 Osiris-Banon systemd[1]: Started Firmware update daemon.
Nov 21 23:41:24 Osiris-Banon systemd[1]: Stopping Firmware update daemon...
Nov 21 23:41:24 Osiris-Banon systemd[1]: fwupd.service: Succeeded.
Nov 21 23:41:24 Osiris-Banon systemd[1]: Stopped Firmware update daemon.

System info:
Ubuntu-Mate 

Linux Osiris-Banon 5.11.0-40-generic #44~20.04.2-Ubuntu SMP Tue Oct 26 18:07:44 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

MrChromebox, 2023-05-17:
TPM 1.2 (and non-CR50 2.0) support added in MrChromebox-4.20.0 firmware release

closed 2023-05-17 by MrChromebox