MrChromebox/firmware #22: Add support for booting off C710's SD card reader

coolstar, 2016-10-25:
Let's add support for the C710's sd card reader. This can be done by adding SdMmcPciHcDxe and SdDxe:

https://github.com/tianocore/edk2/tree/master/MdeModulePkg/Bus/Pci/SdMmcPciHcDxe
https://github.com/tianocore/edk2/tree/master/MdeModulePkg/Bus/Sd/SdDxe

If there's space issues, we can remove the UFS2 and EXT4 filesystem drivers in that order, as they're not used by any boot media. (BSD doesn't even use UFS2 anymore, and Linux distros set up a FAT32 efi system partition, as per UEFI spec)

coolstar, 2016-10-25:
Also, these drivers do work. I tested them by loading them into rEFInd on the C710 and booting the Windows 10 installer successfully from an SD card

ReddestDream, 2016-11-07:
Newest ROMs have added support for booting of C710's SD Card reader.

closed 2016-11-07 by MattDevo