MrChromebox/firmware #24: Issues with Dell Chromebox 3010 & UEFI

dwrz, 2016-10-29:
Tryin go run a headless Ubuntu server on a Dell Chromebox 3010 (i7). I want to be able to reboot the server remotely and have it automatically boot into Ubuntu.

Issues I am having:
1. BIOS menu loads, rather than GRUB.
2. Pressing "Continue" does nothing.
3. Booting from file will run GRUB. Changing the settings to run GRUB and to wait 0/5 seconds also work, but the settings do not save.

ReddestDream, 2016-10-29:
Your issues appear to be a combination of these two known issues:

https://github.com/MattDevo/firmware/issues/4

https://github.com/MattDevo/firmware/issues/10

Fixes for these are planned with an upcoming overhaul of the firmware.

For the time being, to resolve Ubuntu not booting, you can boot an Ubuntu Live USB and run my script here:

https://www.dropbox.com/s/cay7ni1cj3ec88l/UEFIFirmwareLinuxFixScript.sh?dl=0

This will install rEFInd (configured properly; a typical rEFInd install will not work as it, like Ubuntu, doesn't want to overwrite anything Windows might make) to your EFI partition on the internal SSD. This will enable you to boot Ubuntu. It is also possible to manually create the necessary BOOT folder on the EFI partition manually and then move over the necessary files from the Ubuntu folder on that partition, renaming GRUB to BOOTX64.EFI. Hope this helps.

dwrz, 2016-10-29:
Thanks so much for the detailed reply. I reviewed the documentation on mrchromebox and realized that these issues were already mentioned there -- my bad.

I ended up just flashing the legacy BIOS version of the full ROM. Had some issues getting my wireless keyboard working with the Ubuntu server installer, but otherwise everything seems to be up and running now.

Thanks again for your time and help!

ReddestDream, 2016-10-30:
Welcome! ^_^

>I ended up just flashing the legacy BIOS version of the full ROM.

I really do not recommend using the Legacy BIOS ROMs. They are, as the name suggests, Legacy, and they may not continue to be updated after all the issues with UEFI firmwares are fixed, forcing you to have to convert your BIOS-style install or re-install.

dwrz, 2016-11-01:
Thank you! That's good to know -- wish I'd heard about your script before I started setting everything up!

Would I be able to flash the UEFI BIOS and run your script without having to reinstall Ubuntu?

MattDevo, 2016-11-01:
> Would I be able to flash the UEFI BIOS and run your script without having to reinstall Ubuntu?  

yes, but you'd have to do some repartitioning (resize existing data partition, add a small ~100MB EFI partition) and manually install/configure a bootloader like rEFInd.  I also don't have a comparable "headless" option for the UEFI firmware yet, so there'd be no benefit to doing that for your existing setup though, and I'd recommend you stick with the Legacy firmware until it makes sense to switch (eg, going to a dual boot or Windows setup)

closed 2016-11-01 by MattDevo

ReddestDream, 2017-02-25:
@dwrz We have a new UEFI ROM that supports headless. Let me know if you are interested in helping to test. It's not on the main install script yet.

dwrz, 2017-02-27:
Hi -- and thanks very much for your continued work on this.

I'm interested in helping, but don't think I'll be able to for at least another couple of weeks. 

Is that OK?

ReddestDream, 2017-02-27:
We may have it released by then, but that's fine. If we do, you'll get it then. :)

dwrz, 2017-02-27:
Ah, sorry I can't be more helpful. I'll make sure to check in then!