MrChromebox/firmware #25: Unable to build lulu from source

vladimir-lu, 2016-11-07:
I realise this is a bit like asking for unicorns (on a good day) but lulu does not build from source because of missing refcode blob:

```
3rdparty/blobs/mainboard/google/lulu/refcode.elf: No such file or directory
E: Couldn't read in file '3rdparty/blobs/mainboard/google/lulu/refcode.elf'.
```

Unfortunately I have not been able to find a good explanation of what the refcode blob is and where to get it from. (https://review.coreboot.org/#/c/4895/ explains very little). Of course it's available from reading the flash of the board itself but that does not say much about the origin and authenticity of its contents...

Any ideas on how getting the refcode blob can be automated/how it can be validated?

MattDevo, 2016-11-07:
unfortunately, the blobs aren't redistributable (per Intel), so you have to extract them (VGA BIOS, refcode, MRC, ME, IFD)  from one of the following using cbfstool and put them in the lulu 3rdparty/.../lulu folder:
 1) my Lulu ROM 
 2) your stock ROM, read from your device
 3) a shellball ROM extracted from a Google recovery image for Lulu

2/3 will be identical, except for the Intel ME blob, which can't be read from a running system.  1 will have some changes from 2/3:
- VGA BIOS will be newer version (1040 vs 1033/35 IIRC)
- MRC blob is different version, in order to maintain consistency with other Bradwell Chromebooks, and to change the GPU's pre-allocated memory amount from 32MB to 96MB

If you want to build from my tree, you'll need to use either my modified MRC blob, or a stock one from Auron_Paine/Auron_Yuna/Gandof; using a stock Lulu one will brick your device.

vladimir-lu, 2016-11-07:
That's very helpful (especially the comment about MRC blob).

One more question - do all of these blobs come from Intel's FSP? (I also found http://www.intel.com/content/dam/www/public/us/en/documents/guides/5th-gen-core-i5-5350u-eval-kit-fsp-user-guide.pdf but could not find a download of the actual coreboot image - but then again I did not sign up for an account yet)

MattDevo, 2016-11-08:
> do all of these blobs come from Intel's FSP  

no, Broadwell ChromeOS devices do not use FSP (Braswell is the first CrOS platform to do so).  FSP would take the place of the VBIOS, MRC, ME, and likely refcode; you'd still need the firmware descriptor (IFD).  You can't download the stock coreboot image, you have to download the recovery image and extract the shellball ROM (image) from it -- see: https://johnlewis.ie/extracting-the-shell-ball-rom-using-a-chromeos-image/

coolstar, 2016-11-08:
@MattDevo @vladimir-lu also forgot that we have the Intel GOP driver blob, which we extracted from the Insyde h2o UEFI from an Acer E5-571. That's required for graphics on UEFI

MattDevo, 2016-11-08:
@coolstar while true (and actually, my Duet images are using a different/newer GOP driver), the Duet/UEFI payload can basically be considered blob right now, since I haven't cleaned up my edk2 tree to the point to where all targets are buildable via script.  SNB/IVB requiring different commits from HSW/BDW, and Chromeboxes requiring a different vbt (another blob) and timeout value all complicate things a bit

vladimir-lu, 2016-11-09:
For now I'm more interested in reproducing the legacy bios build (as I was struggling to get Qubes to work with the UEFI payload) - but as always the commentary above is informative :)

After extracting the various blobs I was able to build + boot the seabios payload with one strange quirk - the i2c touchpad does not work. It is listed in `/sys/bus/i2c/devices/i2c-SYNA0000:00` but the driver for it (i2c_hid) does not get loaded. This is without touching the config of seabios or coreboot from the defaults provided.

There are some missing features:
* VT-d is not available
* TPM init does not seem to work?

For VT-d after doing some digging it seems like the CPU should support it (I have the i5-5300U version) but the ACPI DMAR table is missing. There is no implementation in the soc directory for broadwell (just for broadwell-d and sandybridge). It _looks_ like that may be an easy thing to add given how similar the two are at a glance, but _looks_ are deceptive when working with hardware.

For the TPM after enabling LPC_TPM it is still listed as disabled but I have not dug into it more deeply than that.

coolstar, 2016-11-09:
The Dell Chromebook 13 does not use an i2c HID touchpad. It uses a proprietary Synaptics RMI HID trackpad.

VT-d is not supported on broadwell by coreboot, and we don't support TPM right now. That may be coming later with newer UEFI firmwares.

coolstar, 2016-11-09:
@vladimir-lu https://www.qubes-os.org/doc/uefi-troubleshooting/

MattDevo, 2016-11-09:
@vladimir-lu for the trackpad to work under linux, either revert commit b7706fb [google/lulu: prevent Windows from loading synaptics I2C driver for trackpad] or add 'acpi_osi="Linux"' to your grub kernel parameter list.  I normally revert this commit for the Legacy Boot Lulu ROM.
I'll have to look at adding VT-d, since I also now have an i5-5300U model Lulu

vladimir-lu, 2016-11-09:
@coolstar - I think I get why `b7706fb` is blocking it from working in Linux by default - Linux doesn't identify itself as itself to ACPI by default and the i2c_hid driver only supports CID "PNP0C50" which isn't set because of the Windows issue described in the commit. It was confusing me for a while because I see the device as `/sys/bus/i2c/devices/i2c-SYNA0000\:00/` in either case but that path must be the HID not the CID. Thanks for the pointers!

MattDevo, 2016-11-09:
@vladimir-lu commit b7706fb was a deliberate choice to work around Windows unsigned driver issues, which had the side effect of breaking the trackpad under Linux without the kernel parameter -- which is why it's not part of the released Legacy Boot ROM/firmware

MattDevo, 2016-11-16:
think we can close this one

closed 2016-11-16 by MattDevo