MrChromebox/firmware #26: Power Management implemented 2016-09-22 seems to cause a hang on boot with some USB hubs

Mal2-ksc, 2016-11-16:
This has been noted by others (ask Reddest) but never before on a wall-powered hub. Well, I found one. This one specifically: http://www.newegg.com/Product/Product.aspx?Item=0J2-008P-00003

It does not ship with a wallwart but I have plenty of 5V 2A power supplies.

Oddly, it started working at some point in the process of trying to unstick the HDMI which had quit when I flashed it.

ReddestDream, 2016-11-17:
I have not experienced this with /any/ powered hubs, but the hubs I have that stopped working don't work for booting on any other PC's either, which is why I considered this a non-issue.

ReddestDream, 2017-01-24:
Not sure this should remain open since it's an issue with hubs rather than the firmware. As noted, this issue occurs on other PCs with some hubs . . .

closed 2017-01-24 by MattDevo