MrChromebox/firmware #34: Non-working External Video on C710 Before OS Load.

ReddestDream, 2016-11-18:
Neither HDMI nor VGA work on IVB/SNB C710 before OS load. They have never worked AFAIK, nor do they work on Stock firmware. They also do not work if you restart from the BMM or from an OS in which they do work. This is in contrast to the BDW issue noted on the C740:

https://github.com/MattDevo/firmware/issues/29

ReddestDream, 2016-12-03:
This is now affects HSW as well after that latest firmware got all the VBTs in sync. :/

MattDevo, 2016-12-03:
@ReddestDream I don't see how that's possible, since my Peppy board works fine w/o an internal display

ReddestDream, 2016-12-03:
I have no HDMI on mine until an OS loads on the latest firmware. Sorry. Perhaps it's a timing thing with my TV like it might be on BDW . . .

ReddestDream, 2016-12-08:
This has now been based on display init priority changes. Issue is now fixed all around. Thanks! ^_^

closed 2016-12-08 by ReddestDream