MrChromebox/firmware #36: Not booting from SD card

pavel, 2016-11-27:
It is a strange issue I must say.

### Environment
* Samsung Chromebook 3 (XE500C13)
* Samsung EVO 32GB Class 10 Micro SDHC Card
* Kodi E-Z Setup Script (RW_LEGACY method)

### To reproduce:
1. Install Arch Linux (2016.11.01) to SD card. Use Syslinux as a boot loader.
I use USB drive to boot ISO in order to install OS to SD card.
I use GPT partitioning. And to install Syslinux I use:
```bash
pacman -S --noconfirm gptfdisk syslinux
syslinux-install_update -iam
```
2. Ctrl-L during Chrome OS warning screen. Esc to view boot options.

### Problem
Card is recognized in the boot menu. Though when selecting SD card booting hangs on "Booting from Hard Drive".

There're several strange things:
1. SD card is bootable. If I burn Arch Linux ISO to SD card it boots without problems.
2. I have exactly the same installation on a USB drive and it boots without problems.
3. Probably worth mentioning that when installing Syslinux (USB or SD card) I get an error that Syslinux failed to install boot flag on a drive. Though it does not affect USB booting at all.

---
PS: Is there any way to debug what is going on during this boot hang?

MattDevo, 2016-11-27:
which version of syslinux?  I know 6.0 thru 6.0.3 are problematic on older ChromeOS devices w/stock firmware; 6.0.4-pre1 works fine (as does the older 5.10)

pavel, 2016-11-27:
I've managed to get it working with GRUB (MBR and GPT partitioning). So I guess we can agree that it is Syslinux issue if you don't see anything wrong with the firmware.

MattDevo, 2016-11-27:
I'm familiar with that model, I just haven't tested it to know if it has issues with syslinux 6.0.3 as well, but seems like it might.  Given that the SD card shows up and boots with Grub, I can't see this being a RW_LEGACY firmware issue

closed 2016-11-27 by MattDevo

pavel, 2016-11-28:
FYI I managed to get Syslinux working. I did manual install which is mentioned in Arch Linux wiki [here](https://wiki.archlinux.org/index.php/syslinux#GUID_partition_table).