MrChromebox/firmware #38: Reboot on Resume From Suspend - Full Rom Bay Trail

natloh, 2017-01-05:
When using the full rom on a bay trail device, resuming from suspend results in a hard reboot.
Best workaround I have found is to disable suspend all together and so you shutting off the screen when the lid is closed. Not ideal, but it helps conserve a little battery.

ReddestDream, 2017-01-05:
We are aware of this issue and working on finding the cause. Might be improper handling (read: no handling) in Coreboot for the Intel Management Engine.

ReddestDream, 2017-01-19:
I believe I have found a solution to this issue. The write-protect range is not being properly cleared upon flashing, and this is breaking the MRC cache in custom firmwares. This should be fixable by clearing the write protect range before flashing (flashrom --wp-range 0 0).

ReddestDream, 2017-01-24:
This is now fixed in the 01/23 UEFI Bay Trail ROM.

closed 2017-01-24 by MattDevo