MrChromebox/firmware #42: Software Sync or some other EC Firmware Update Method.

ReddestDream, 2017-01-14:
I have an EC firmware update ready and tested on HSW, BDW, and BYT that makes the keyboard on Chromebooks properly PS/2 compliant. This allows the stock Windows PS/2 keyboard driver along with any other keyboard drivers that are not aware of the  . . . peculiarities (read: noncompliance) . . . of ChromeEC keyboards to work properly. With this, the keyboard will now work in the Windows installer and with basic functionality in Windows itself (CoolStar's driver is needed for media key support).

I also have DSDT changes to facilitate stock keyboard functionality as well as support for CS's driver for SNB/IVB/HSW/BDW/BYT. The only missing piece is a mechanism for updating the EC firmware, be it in the firmware itself a la Software Sync or through some sort of menu option. The DSDT changes are fine w/o the EC firmware update, but HSW/BDW/BYT will not be able to use the stock Windows keyboard driver w/o my EC firmware fixes (HSW needs the cherry-picked retry fix + my Fix; BDW/BYT only need my fix). Thx.

ReddestDream, 2017-01-16:
@MattDevo notes: "so, finally managed to piece together all the bits needed for EC software sync"

Matt Devo, [Jan 15, 2017, 15:03]: 
https://chromium.googlesource.com/chromiumos/platform/vboot_reference/+/firmware-lulu-6301.136.B/firmware/lib/vboot_api_kernel.c#629

https://chromium.googlesource.com/chromiumos/platform/depthcharge/+/firmware-lulu-6301.136.B/src/vboot/callbacks/ec.c#256

https://chromium.googlesource.com/chromiumos/platform/depthcharge/+/firmware-lulu-6301.136.B/src/drivers/ec/cros/ec.c#734

Paviluf, 2017-02-05:
Will it be possible to have more reasonable value for the fan speed than the default one for Falco (and maybe others)? The default are insane, far too high with 2700 rpm in idle as you can see [here](https://chromium.googlesource.com/chromiumos/platform/ec/+/firmware-falco_peppy-4389.B/common/thermal.c). I had to use some ectool / systemd trickery to keep the noise low.

For the record the fan rpm values I use are these one :

TEMPS=(45 55 65 70 75)
FAN_SPEEDS=(1300 1600 1900 2200 2700)

ReddestDream, 2017-02-05:
I don't have a Falco to test on, and I don't have any devices with an overly noisy fan to test changes on . . .

I might be able to make you something to test tho later. Are you willing to risk some damage?

Paviluf, 2017-02-05:
Well I love to test things but since I use the chromebook for my developer training (school) I a bit reluctant.

Is the risk high or low ? :D

ReddestDream, 2017-02-05:
Maybe 30%? IDK. Probably not bricked, but it might behave weirdly and potentially corrupt data until you reboot. Also, when I tweak the fan profile that might cause things to overheat.

I need you to run "ectool version" and tell me what EC versions Falco has. I'm not 100% sure it share some versions with Peppy. The notion in the Chromium source is unclear . . .

ReddestDream, 2017-02-05:
I would really appreciate having another HSW book to test the keyboard changes on as well, so I would put that in too . . . I just need to know the commit info for the EC firmware it uses (should be something like "5d52788") so I can be sure I use the right source.

Paviluf, 2017-02-05:
Here is the "ectool version"
```
RO version:    falco_v1.5.132-c77d95f
RW version:    falco_v1.5.132-c77d95f
Firmware copy: RW
Build info:    falco_v1.5.132-c77d95f 2014-12-09 12:12:39 @build121-m2
```

ReddestDream, 2017-02-05:
Okay. Good. It should be able to take the same EC firmware as Peppy.

Can you debrick with a SOIC-8 clip? We are only going to flash the RW part of the EC, so, worst case scenario, you will need to externally flash a new main firmware since the current one auto-boots in to RW. If you don't, I can ask @MattDevo to make you a special test firmware that doesn't reboot the EC to RW automatically so we can be 100% sure before you go back to the current release version.

Paviluf, 2017-02-05:
> Can you debrick with a SOIC-8 clip? 

I don't have a SOIC-8 clip so no.

> If you don't, I can ask @MattDevo to make you a special test firmware that doesn't reboot the EC to RW automatically so we can be 100% sure before you go back to the current release version.

That will be awesome ! I really should not break it since I must have a working laptop for school ;)

ReddestDream, 2017-02-05:
Okay. I made you the EC firmware to test, and I've contacted @MattDevo.

I'm about 80% sure there will be no issue. I'm only really concerned about these fan changes and the fact that Falco has three separate EC firmwares branches. Your info told me which to use tho . . .

Paviluf, 2017-02-05:
Thank you very much ! Don't worry about the fan changes. I tested the chromebook without the fan for some time and it never was too hot. I hope I didn't do any mistake about the EC firmwares branches :D

ReddestDream, 2017-02-08:
Are you around? We have the RO ROM ready on the test script for you? 

Can you come on the /r/chrultrabook Discord? Or can we chat some other way?

Paviluf, 2017-02-08:
No problem to chat on discord, I just need to create an account. I will be available at 6:30pm utc.

ReddestDream, 2017-02-08:
I'll try to be around then, but no guarantees. Message me directly.

Here's the chrultrabook Discord: https://discord.gg/tkPTk5w

MattDevo, 2017-02-21:
implemented in 2017.02.21 release

closed 2017-02-21 by MattDevo