MrChromebox/firmware #44: HDMI Scaling Issue in Windows on BYT.

ReddestDream, 2017-01-24:
On BYT (tested on SQUAWKS), Windows does not scale properly over HDMI until 1360x768 is enabled and Maintain Aspect Ratio is set in the Intel Driver. This is odd because on other models, proper scaling occurs automatically. Still looking for confirmation on other BYT models . . .

coolstar, 2017-01-24:
Nothing to be done in the firmware. The VBT has the values set correctly. Once Boot Services are exited, the Intel driver in the OS takes over and the EFI GOP driver stops running. This is handled in the bay trail intel graphics driver.

closed 2017-01-24 by MattDevo

ReddestDream, 2017-01-24:
I might post this one again if I get more info from other BYT books.

ReddestDream, 2017-01-27:
Seems it doesn't affect 1080p BYT books. :/