MrChromebox/firmware #46: Get Rid of Legacy SeaBIOS ROMs.

ReddestDream, 2017-01-26:
The Legacy SeaBIOS ROMs are buggy, untested, and unnecessary. They keep Linux users on out-of date firmwares, and they perpetuate the idea among Linux users that the UEFI firmwares are the "Windows" firmwares when this is not true. Linux works fine on the UEFI firmwares. The only limitation is that Debian/Ubuntu-based distros need a workaround after installation:

https://github.com/MattDevo/firmware/issues/4

https://github.com/MattDevo/firmware/issues/10

See here for the latest SeaBIOS ROM issues:

https://github.com/MattDevo/scripts/issues/24

I have written a script to help Debian/Ubuntu-based distro users fix booting, and I am more than happy to write various guides to work around the installation issues on the UEFI firmwares. Since we have the firmware UEFI Shell in place now, it is actually possible to fix the installation issues from that shell by moving and creating the correct files and folders. Having Legacy SeaBIOS ROMs just to facilitate Linux installation is excessive and unwarranted. On Books, they have no purpose that justifies their existence . On Boxes, there is some purpose because of the issues with headless operation and detection failures on some displays, but even then they should not be generally available and we should actively be investigating fixes for the UEFI firmwares. Thx.

reynhout, 2017-01-29:
Please keep in mind that removing legacy support from full firmware would effectively break the upgrade path for tens of thousands of existing users.

A migration script would help mitigate the pain, but it would still create a significant support load with ambiguous benefit to the user.

If SeaBIOS has become an overwhelming burden to maintain, one way forward is to deprecate and document. Describe the advantages of UEFI over BIOS booting, and provide clear instructions for getting there, but don't force people to make decisions they don't understand.

Without legacy support, switching to full firmware would be a bad suggestion for most existing users. We would be asking users (of varying skill levels) to perform an unfamiliar manual process, where the consequences of error or confusion lead to a non-bootable system, complicated recovery, and angry support requests. The advantages gained would have to be very large to outweigh that risk.

ReddestDream, 2017-01-29:
>Please keep in mind that removing legacy support from full firmware would effectively break the upgrade path for tens of thousands of existing users.

Users who choose RW_LEGACY do so because they either need ChromeOS or they cannot, for whatever reason, risk opening the device to remove the screw for fix the jumper to update the firmware. Users really should be making be making the decision about RW_LEGACY vs chrx/Crouton vs FULL ROM at the beginning and understand that if they need to change between, reinstallation will be necessary.

 In addition, we have had success getting ChromeOS working on the UEFI firmware, but the process is currently complex and breaks autoupdating and channel switching due to the integration of ChromiumOS components:

https://github.com/MattDevo/firmware/issues/35

>A migration script would help mitigate the pain, but it would still create a significant support load with ambiguous benefit to the user.

The migration script already exists. I made it months ago and had several users test it. I was told that it was a kludge and bad, risky solution that could easily break. I admit this and intended it as a temporary solution. But no one else has even given it much thought . . . :/

>If SeaBIOS has become an overwhelming burden to maintain, one way forward is to deprecate and document. Describe the advantages of UEFI over BIOS booting, and provide clear instructions for getting there, but don't force people to make decisions they don't understand.

The major problem, as I mention above, is that is unrealistic to keep them in sync anymore. The implementation of OpRegion in Coreboot, necessary for the UEFI ROMs, has changed how the graphics work and made it difficult to keep the Legacy ROMs, which need different configuration in Coreboot, working properly. We saw this with the Backlight issues recently. It's not just a matter of compiling with a different payload anymore. The UEFI ROMs since 12/14 no longer even need SeaBIOS as a shim, which has sped up performance and enabled us to implement more features for them.

In addition, maintaining two ROMs for each device effectively doubles the testing time, which already consists of testing 3 OSes in multiple versions. I test Windows 8.1, Windows 10, Linux (kernels 4.4, 4.7, and 4.10) as well as a macOS 10.11 and 10.12 on supported models (HSW/BDW) across five generation: SNB, IVB, HSW, BDW, and BYT.

As to advantages, the advantages regarding Windows and macOS I don't think interest you, so I will not discuss them here. If you are interested, let me know.

As to Linux, it is true that Linux does not benefit as much as Windows and macOS from the transition to UEFI. I'm not going to deny that. Part of the reason for that is that development time has not been spent there since Linux users have had the Legacy ROMs. Problem now is that we have reached the breaking point where maintaining two sets of firmware is now a good deal harder. Before, at least the UEFI firmwares still depended on SeaBIOS as shim and the VBIOS to provide graphics support. Since that is no longer the case, the ROM types have now diverged sufficiently that Coreboot changes for one can now break the other . . . :/

There is also the unfortunate decision that Debian made and Ubuntu inherited to use NVRAM to change the boot priority on UEFI firmwares in order to boot. NVRAM support does not exist in the Coreboot UEFI package (CorebootPayloadPkg). We would have to implement it, and this will take time.

This does not mean that Linux doesn't work or is somehow less supported on the UEFI firmwares. On the contrary, I test Linux, multiple kernels, with each UEFI firmware release. Also, as you know GalliumOS has been fixed to account for this NVRAM issue. Non-Debian-based mainstream distros also do not have this issue, as they made the decision to handle UEFI booting more like how Windows does it. This issue also affects other non-Chrome PC hardware where the vendor left out NVRAM as a cost saving measure as Windows does not make much use of it.

See this these issues:

https://github.com/MattDevo/firmware/issues/4

https://github.com/MattDevo/firmware/issues/10

On this point of forcing people to make decisions they don't understand, let's all step back and realize that users are getting this for free and no one is ever forcing them to update the firmware or change firmware types. This point about user choice has to be balanced against developer time and keeping users up-to-date. We would actually love to have even more feedback from Linux users using the UEFI firmware, and we have quite a few such users on /r/chrultrabook.

There just has been this notion going around that the UEFI ROMs are the "Windows" firmware, and this is actually painful to me given the amount of testing on Linux I do. Other than the NVRAM issue, on SNB/IVB/HSW/BDW/BYT books, Linux support is at parity with Windows support on the UEFI firmwares if not exceeding it (e.g., Audio on BYT due to the missing driver for Windows). Good compliant firmware can support a variety of OSes with minimal workarounds. For example, in case you didn't know, OSI=Linux is no longer necessary on any device. We found a better way for Lulu's touchpad that makes both Windows and Linux happy at the same time. Just took more reading through ACPI manuals and kernel source . . .

[... cut ...] ssing need to. And if they decide to, I don't think it's that hard to provide some documentation for the process (which I could write up and others could edit and help with) of preserving existing installs, which the caveat that they should make proper backups and eventually just reinstall to ensure maximum stability.

The advantages of UEFI for Linux users will unfortunately have to come /as/ users transition to it, rather than before, which was the same in the case of our Windows support . . .

closed 2017-04-09 by ReddestDream