MrChromebox/firmware #47: Battery notifications

Paviluf, 2017-01-28:
Hello,

I was wondering if these power (battery) notification (more infos in the link) can be induced by the firmware.

https://bugzilla.gnome.org/show_bug.cgi?id=777611
[image]

I'm still on a legacy (up to date) version (I plan to update to an uefi one soon when I will reinstall my OS (Fedora)) 

Thanks !

ReddestDream, 2017-01-28:
I don't think these are due to anything in the firmware, but you need to update to the UEFI firmware (which is actively developed) for us to be sure.

Paviluf, 2017-01-28:
I will update to the uefi firmware but it will take some time. I want to upgrade the ssd before installing Fedora 25 and Windows 10 but as I use the chromebook for my developer training (school) I'm a little bit stuck :)

ReddestDream, 2017-01-28:
Okay. Can you close this issue for now and at least wait until you can test on the UEFI firmware?

I am 90% sure tho that this issue has nothing to do with the firmware and is just an OS configuration issue.

Paviluf, 2017-01-28:
Ok I close this issue for now. I will test on the UEFI firmware and report back.

It's funny because the Gnome dev said that it's a firmware (or hardware) issue and you say that's a software issue :D

closed 2017-01-28 by Paviluf

ReddestDream, 2017-01-28:
Thx.

>It's funny because the Gnome dev said that it's a firmware (or hardware) issue and you say that's a software issue :D

IDK. Could be both. But the battery is fine on other distros. It's something we've worked on and tested.

Paviluf, 2017-02-05:
It seems that is the normal behaviour of gnome. It seems that is the "computer will suspend very soon because of inactivity" notification. I will seek confirmation on [gnome](https://bugzilla.gnome.org/show_bug.cgi?id=777611).