MrChromebox/firmware #48: When the chromebook is in standby and you connect a usb device, the usb device light up whereas it should not.

Paviluf, 2017-02-05:
Hello,

As I say in the title, when the chromebook is in standby and you connect a usb device, the usb device light up whereas it should not.

Obviously not a very important bug but I just wanted to let you know :)

Thanks !

Paviluf, 2017-02-05:
Oh I forgot to say that I have a Falco with the latest uefi full rom on Fedora 25 ;)

ReddestDream, 2017-02-05:
If you let it sit in standby for a bit, that should stop. Sometimes it takes a bit for the USB ports to power down.

Paviluf, 2017-02-05:
Ok I will try :)

ReddestDream, 2017-02-11:
@Paviluf Are you also ready to close this one?

Paviluf, 2017-02-11:
No I need to do more tests but it seem that the usb port doesn't completely switch off when the chromebook is in standby. For example, when I switch the chromebook to standby with a usb mouse attached, the usb mouse switch off and that's perfect. But if I connect an other usb device (or if I unplug / plug the usb mouse) to the chromebook, this usb device will switch on.

ReddestDream, 2017-02-11:
So, when you say standby, you are talking about a sleep state as opposed to a hibernate state?

That USB behavior is normal in sleep but not in hibernate.

Paviluf, 2017-02-11:
I mean when you close the lid

ReddestDream, 2017-02-11:
Just closing the lid is normal sleep (S3), so my comment above applies.

Paviluf, 2017-02-11:
You seem to be right ! I thought the usb ports should be off in standby, since my usb mouse was off.
Thanks !

closed 2017-02-11 by Paviluf