MrChromebox/firmware #50: HP Chromebook 11 G5 braswell CTRL+L goes to BlackScreen

johntwillis, 2017-02-08:
Hi,

I have an HP Chromebook G5 with a braswell Celeron N3050 cpu with Intel HD 400 graphics

P/N X9U05UT#ABA

The script seems to identify it and succeeds in applying the RW_Legacy update, but when I go to reboot, the Developer mode boot screen comes up.. I type CTRL+L and it wipes the chrome bitmap screen down to a blank black screen and never produces a display.

I thought it was suppose to produce an ESC menu and detected options for boot.

I got this in November so I though the eMMC or USB devices might be so new they aren't properly identified. Do you have any suggestions, or is there any data I could collet and contribute to help solve the problem? Thanks

ReddestDream, 2017-02-08:
AFAIK, this is due to a ChromeOS bug after a recent firmware update. Still looking into it.

MattDevo, 2017-02-08:
correct - seems to be ChromeOS breaking things.  See: https://github.com/GalliumOS/galliumos-distro/issues/320

closed 2017-02-08 by MattDevo

johntwillis, 2017-02-08:
That is confirmed, thank you MattDevo.

I followed the link, and downloaded the SETZER 8530.93.0 recovery image, then used the ChromeOS Recovery app to write that to an USB thumbdrive and restarted into recovery mode. That did powerwash my personal settings and the files that were on the internal eMMC storage (as predicted).

After the recovery I followed the directions to remove the USB thumbdrive and it automatically rebooted, and came to rest at the white dev screen waiting for a CTRL+D - instead I used CTRL+L and that proceeded to wipe the screen black, then produced a stream of white on black text that said SeaBIOS. then went black again. (I did not have any USB devices plugged in right after the recovery.. but I was encouraged).. I reinserted USB boot media and an SD card.. and power cycled. This time the SeaBIOS did the same but also cleared the text and produced [Press ESC for boot menu.] which I did and it listed three choices:

1. MMC drive SW16G 15028MiB
2. SD card SD32G 30700MiB
3. USB MSC Drive Generic USB  SD Reader 1.00

Pressing 1. produced a Black screen and did not boot into Chrome.
Pressing 2. was to no effect because the card is empty
Pressing 3. booted to Windows 10 install image (I didn't know what I had on it) then it crashed, lol

Lots more fun.

I can't wait till support gets better.

ReddestDream, 2017-02-08:
BSW support is being worked on. I got test devices . . .