MrChromebox/firmware #51: peppy: keyboard input not detected in systemd-boot

Zirkelite, 2017-02-13:
When using systemd-boot as UEFI bootloader, systemd-boot does not detect keyboard input when using the integrated keyboard. However, an external USB keyboard works as expected. Other bootloaders such as GRUB work correctly with the integrated keyboard.

This is potentially a firmware issue, since systemd-boot works on other computers with different keyboard combinations as expected.

ReddestDream, 2017-02-13:
Yeah. I've also noticed that. I'm expecting it will be fixed once the EC firmware keyboard fix is in to make the keyboard properly PS/2 compliant . . .

https://github.com/MattDevo/firmware/issues/42

ReddestDream, 2017-02-14:
Okay. This is odd. I tested out systemd-boot and even with the keyboard EC firmware fixes, it still doesn't work in systemd-boot. Something we will have to look into.

ReddestDream, 2017-02-20:
Figured out why this is happening. Should be fixed in the next major update.

https://github.com/ReddestDream/chrome-ec/commit/4759fa2e640368c87145df4da9575d04fa247b78

MattDevo, 2017-02-21:
fixed via EC firmware update in 2017.02.21 release

closed 2017-02-21 by MattDevo