MrChromebox/firmware #54: Bay Trail Audio non-functional

coolstar, 2017-03-29:
The Bay Trail Audio Codec has missing GPIO data, resulting in a Code 10 error.

Also the GPIO's are incorrectly under LPEA, causing a Code 12 error when placing GPIO's in the Codec.

GBUF should be empty and CODC should be replaced with this:

```Device (CODC)
        {
            Name (_HID, "193C9890")  // _HID: Hardware ID
            Name (_DDN, "Maxim 98090 Codec")  // _DDN: DOS Device Name
            Name (_UID, One)  // _UID: Unique ID
            Name (_CRS, ResourceTemplate ()  // _CRS: Current Resource Settings
            {
              GpioIo (Exclusive, PullNone, 0, 0, , "\\_SB.GPSC", 0, ResourceConsumer, ,){
                0x000E
              }
              GpioIo (Exclusive, PullNone, 0, 0, , "\\_SB.GPSC", 0, ResourceConsumer, ,){
                0x000F
              }
                I2cSerialBusV2 (0x0010, ControllerInitiated, 0x00061A80,
                    AddressingMode7Bit, "\\_SB.PCI0.I2C2",
                    0x00, ResourceConsumer, , Exclusive,
                    )
                Interrupt (ResourceConsumer, Level, ActiveLow, Exclusive, ,, )
                {
                    0x00000043,
                }
            })
            Method (_STA, 0, NotSerialized)  // _STA: Status
            {
                If ((S2EN == One))
                {
                    Return (0x0F)
                }
                Else
                {
                    Return (Zero)
                }
            }
        }

xDiiThy, 2017-03-29:
Hi, how would I implement this fix on the Toshiba Chromebook 2 Swanky, I have the latest UEFI

MattDevo, 2017-03-29:
cmon @coolstar, you know a pull request is the correct way to handle this

closed 2017-03-29 by MattDevo

xDiiThy, 2017-03-29:
So yah i'll just wait for an update here, thanks will be donating as soon as its all setup remarkable work!