MrChromebox/firmware #55: Possible boot issues

Paviluf, 2017-03-29:
Hello,

I see these 3 messages everytime my Falco boot with Fedora 25. I don't know if it's Fedora problems or firmware problems.

> System BootOrder not found. Initializing defaults.
> Creating boot entry "Boot0002" with label "Fedora" for file "\EFI\fedora\shim.efi"

[image]

> EFI stub: ERROR: Could not determine UEFI Secure Boot status.

[image]

> usb 1-4: string descriptor 0 read error: -22

[image]

Thanks !

ReddestDream, 2017-03-30:
The first error is because the firmware doesn't support boot orders yet.

The second error is because the firmware doesn't support Secure boot.

The third error is caused by a bad USB device name in the Bluetooth part of the Wifi device. Can't be fixed.

Hope this helps! ^_^

Paviluf, 2017-03-30:
> Hope this helps! ^_^

Yes ! Thank you !

closed 2017-03-30 by Paviluf