MrChromebox/firmware #56: Toshiba Chromebook 2 (Gandof) i3 - UEFI help

dalbert2, 2017-03-30:
I installed the preferred solution (firmware replacement) for my Toshiba Chromebook 2 CB35-3350.  Now, the system boots to the Mr. Chromebox (03/19/2017) shell which I can exit and access the menu, but when I insert a USB flash drive with the Gallium ISO written to a USB flash drive (tried Etcher, Win32DiskImager and Rufus) and the USB flash drive is not seen and will not boot.  I've tried with the 2.1 Gallium ISO and 18.1 Linux Mint.  The boot manager does not see the USB flash drive.

I seem not to be the only one with this issue; any guidance on how to get an OS installed with this combination (Toshiba Chromebook 2 CB35-3350 with Mr.Chromebox firmware 03/19/2017) would be very much appreciated!

ReddestDream, 2017-03-30:
Two ideas:

1. Try booting with the USB drive in, getting to the menu, then selecting "Boot Maintenance Manager" and "Reset System." This will reboot the system with the USB drive still in. Sometimes, this gives the system more time to find it.

2. Try another USB drive. There seem to be some USB drives that are just too slow for the firmware to properly detect.

dalbert2, 2017-03-30:
Thanks Reddest, you're the best!

I had already tried many variations on idea 1, but idea 2 did the 
trick!  When I switched to a Kingston DT Micro flash drive (only USB 
2.0), it booted right up...I'm installing Gallium now.

Best regards,
Dave

ReddestDream, 2017-03-30:
Welcome! \^_^

Please close this issue. Thx.

closed 2017-03-30 by dalbert2