MrChromebox/firmware #58: UEFI for Chromebook Pixel (2013)?

kevinfessler, 2017-04-10:
Will the 2013 version of the Chromebook Pixel support UEFI? I have a Chromebook I can work with here and I would love to get Windows 10 working on it.

MattDevo, 2017-04-10:
While I'd like to have UEFI firmware for the 2013 Pixel just for completeness, it's unlikely to see full support in Windows due to unique design considerations and lack of drivers for hardware found only on that device.

kevinfessler, 2017-04-11:
I wonder what work could be done in order to get the keyboard and trackpad to work? I wouldn't mind legacy boot as long as those two were functional. What is the difference between the 2015 and 2013 keyboard/trackpad drivers in Windows? Thanks for your help! 😉

MattDevo, 2017-04-11:
ask coolstar on https://www.reddit.com/r/chrultrabook what's needed to get those working.  The keyboard should work under Windows with his current drivers, but think the trackpad is likely a lost cause due to being routed thru the GPU for some bizarre reason

kevinfessler, 2017-04-11:
Sounds good, I'll look into it. Thanks for your help!

closed 2017-04-11 by kevinfessler