MrChromebox/firmware #62: Power on AC for Acer C710  (Boot after power outage)

ShapeShifter499, 2017-04-20:
I was recently reading https://plus.google.com/104852207441570817808/posts/HTi8t4jf1MW   and I was wondering if something similar could be done for the Acer C710 Coreboot.  Can Wake-on-Lan be enabled?  Or power on AC?

 I want to use my Chromebook as a NAS server with Arch Linux and Nextcloud.   The issue is finding a way to power the device back on when power is lost.  I was looking at Wake-on-Lan but I couldn't even get it working in Sleep/Hibernate mode.

ReddestDream, 2017-04-20:
AFAIK, I do not think the C710 hardware supports WOL or power on AC w/o the power button . . .

ShapeShifter499, 2017-04-20:
@ReddestDream ethtool from Arch Linux repositories seems to think it's a feature, but it doesn't work.  I think it's a BIOS issue but I have little idea how to change that. 

```
root@LEDE-Main:~# /usr/bin/etherwake -u
usage: ether-wake [-i <ifname>] [-p aa:bb:cc:dd[:ee:ff]] 00:11:22:33:44:55

	This program generates and transmits a Wake-On-LAN (WOL)
	"Magic Packet", used for restarting machines that have been
	soft-powered-down (ACPI D3-warm state).
	It currently generates the standard AMD Magic Packet format, with
	an optional password appended.

	The single required parameter is the Ethernet MAC (station) address
	of the machine to wake.
	The MAC address may be found with the 'arp' program while the target
	machine is awake.

	Options:
		-b	Send wake-up packet to the broadcast address.
		-D	Increase the debug level.
		-i ifname	Use interface IFNAME instead of the default 'eth0'.
		-p <pw>		Append the four or six byte password PW to the packet.
					A password is only required for a few adapter types.
					The password may be specified in ethernet hex format
					or dotted decimal (Internet address)
		-p 00:22:44:66:88:aa
		-p 192.168.1.1

root@LEDE-Main:~# /usr/bin/etherwake -D -i eth0 20:89:84:C1:68:56
The target station address is 20:89:84:C1:68:56.
Packet is  20 89 84 c1 68 56 20 89 84 c1 68 56 08 42 ff ff ff ff ff ff 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56 20 89 84 c1 68 56.
Sendto worked ! 116.
root@LEDE-Main:~# 
```

ShapeShifter499, 2017-04-20:
Oops posted the wrong log from a IRC chat earlier, correct ethtool output is above now.

ReddestDream, 2017-04-20:
The wifi card and the firmware have to work together to support that packet.

What I mean is, I think the wifi card the C710 comes with stock (AR9462) can support it, but I'm not sure the firmware does.

ReddestDream, 2017-04-20:
So, I've looked into the code.

Few points:

1. There is no magic switch to just turn WOL on in the firmware for the C710. For some devices, there's an option. Here, there is not. :/

2. WOL is partly dependent on the Embedded Controller. For Chromebook prior to Link, there was not a standardized Embedded Controller and their capabilities varied. The C710 is prior to Link, so it's possible the Embedded Controller simply doesn't support WOL.

ShapeShifter499, 2017-04-20:
@ReddestDream I was trying thw 'wol' package from Arch Linux and 'etherwake'  from OpenWrt/Lede. I couldn't get either to kick start the Chromebook from both full power off and Sleep/Hibernate.   'etherwake' mentions something about a 'password' for some ethernet devices, do I need that in this case?   Is it possible this card looks for something non-standard on the network?

ReddestDream, 2017-04-20:
I don't think so. I think this is simply a case where, while the networking hardware (the ethernet and wifi themselves) would support WOL, the firmware doesn't have the code for it, likely because the Embedded Controller doesn't support it and doesn't get or at least doesn't respond to the WOL packet at all.

ShapeShifter499, 2017-04-20:
@ReddestDream what about powering on on AC detection?

ReddestDream, 2017-04-20:
That's also Embedded Controller territory. I'm not sure how to change it or if it is possible on pre-Link devices since it was not standardized.

ShapeShifter499, 2017-04-20:
@ReddestDream I'm sorry, what is "Link"?   I never heard of that term in this context before, some sort of standardization?

ReddestDream, 2017-04-20:
Link is the Chromebook Pixel 1. It was the first Chromebook to use the GoogleEC embedded controller with standardized EC firmware. We are able to change/update the GoogleEC's firmware, so for devices after Link, we have a lot more control over low-level behavior.

ShapeShifter499, 2017-04-20:
Oh how cool.

So do you have any other ideas about booting after a power outage?

ReddestDream, 2017-04-20:
More info on Embedded Controllers:

https://en.wikipedia.org/wiki/Embedded_controller

>So do you have any other ideas about booting after a power outage?

I'm not sure. From my experience with C710 boards, they require a power button press to turn on and there is no obvious way of shorting that. :/

ShapeShifter499, 2017-04-20:
@ReddestDream I guess I'll just have to setup some scripts to safely stop anything writing data and power down drives connected via USB in the event of a power loss on AC. If AC power is restored before critical battery level then restart services (i.e. Nextcloud).   I wish there was a better solution, this is the only spare device I have besides my main Laptop where I put so much RAM into, 12Gb total. 

Thank you for the help and information. I'll leave this open in case anyone with the same Chromebook has a similar issue OR in case someone actually knows a workaround.

ReddestDream, 2017-04-20:
So, I've done more review.

I have an idea that /might/ get wake-on-lan/wake-on-lan-over-wifi to work.

I'll have to try to implement something in the firmware. And then have you test it with your config if I can get some progress here.

But I will need you to be on the latest UEFI firmware. This fix will only be in the UEFI firmware if I make it.

If you are on a Legacy SeaBIOS firmware and you upgrade to UEFI, you will lose your ability to boot any BIOS-style Linux installations you may have. If re-installing is too much of a pain given the complexity of your config, then it's not going to be worth me trying to implement this idea I've had since you are the only person who's requested it.

Does this make sense? Let me know. Thx.

ShapeShifter499, 2017-04-21:
@ReddestDream Before I saw your latest post I just went and updated to the latest firmware I could get from @MrChromebox just to make sure I had the latest fixes and whatnot. I spent some time and after reading through the Arch Linux Wiki I was able to convert my current install into a UEFI install.

I have three partitions, boot, root, and swap.   I had to reformat the old boot partition to a fat partition that had the EFI boot flag. I have "boot" mounted to /esp and then I have '/esp/EFI/arch' bind mounted to '/boot' in the file system. According to the Wiki it's the best way to allow pacman to continue installing kernel upgrades as normal while allowing EFI to boot a grub install from that partition. 

TL;DR I have everything booting after only having to reinstall the boot stuff.  For anyone reading this works only because Arch Linux enables support in their kernels to load from a fat partition.  I ran the script from 'https://mrchromebox.tech/firmware-util.sh' so I should be on the latest firmware there is.

ReddestDream, 2017-04-21:
@ShapeShifter499 Cool. I'll take a look at enabling wake-on-lan/wake-on-lan-over-wifi in the firmware with a trick I discovered a few days ago . . .

ShapeShifter499, 2017-04-28:
@ReddestDream @MattDevo Any updates on this?

ReddestDream, 2017-04-28:
Been busy. Have not had a chance to look into it. Still on my list.

ShapeShifter499, 2017-05-08:
@ReddestDream I'm not trying to rush anything just wondering were you are on this.

ReddestDream, 2017-05-08:
I have not had time yet. But it is on my list to work on.

I can't guarantee anything, and I will let you know if/when I'm able to come up with something.

ShapeShifter499, 2017-06-01:
@ReddestDream   Updates?

ReddestDream, 2017-06-01:
Wish I had updates. Sorry. I'll work on it after my finals. :/

darkknight1812, 2017-07-11:
@ReddestDream Any news/updates?

ReddestDream, 2017-07-11:
Not planned for the next update, but I will take a look for the next one.

ShapeShifter499, 2017-09-09:
@ReddestDream any progress yet?

ReddestDream, 2017-09-10:
So, I have taken a look at it, and it seems it's more complex than I thought . . .

WoL on Linux is not something I really understand . . . I mean, I've never used WoL much generally.

We are still a way from the next firmware release.

Maybe I'll be able to get something for you to test before then . . .

ShapeShifter499, 2017-12-09:
I've been running this as a hobby file server right now and I've  set up everything for a graceful shutdown.   The power on when away is the only big issue for me with this set up.

@ReddestDream any updates?

ReddestDream, 2017-12-24:
I might have a chance over the holidays . . .

ShapeShifter499, 2018-05-05:
I still would like to see this feature and would happily reuse my Acer Chromebook C710 as a server but right now I'm planning on converting my C710 back into a standard laptop. I have since switched to using a Raspberry Pi 3B+ for all of my server needs including a Nextcloud, Syncthing, and Wallabag install.   I was originally worried about the RAM requirements but it seems to handle all of my personal stuff just dandy.

ShapeShifter499, 2018-05-05:
I'll leave this issue open for anyone else who wants or needs this feature.

ShapeShifter499, 2018-06-05:
@ReddestDream @MattDevo I'm going to close this issue in a week if there is absolutely no possibility of getting something working here.

darkknight1812, 2018-06-06:
I did find a work around for this in Windows. I have been using an usb ethernet adapter that supports WoL. You have to mess with the power settings for the usb ports in device manager to get it to work. I mainly use the laptop for Kodi and Xbox Game Streaming and this work around works most of the time (if the os updates/restarts, having the lid closed sometimes causes the laptop to shutdown instead).

ShapeShifter499, 2018-06-06:
@darkknight1812 this won't work for my use case  I don't think.    I will be running Linux on this laptop for some server programs.  The problem is not being able to remotely turn on the laptop in the event of a power outage.

MattDevo, 2018-06-06:
@ShapeShifter499 are you sure there's a difference between the stock firmware and mine?  Power-on behavior is handled by the EC, which unlike newer devices, isn't made by Google / open source.  I don't handle anything differently w/r/t EC init or power settings.  

Given your use case, a Chromebox would be a much better fit, as those can be configured to turn on automatically any time AC power is applied

ShapeShifter499, 2018-06-08:
@MattDevo I could have sworn I talked with you or someone on IRC @ freenode on the #coreboot channel.   With stock firmware the Acer C710, the laptop would power itself on automatically if it was off when a AC cable was plugged in.  After installing the modified firmware it no longer does this.

ShapeShifter499, 2018-06-11:
@MattDevo @ReddestDream @darkknight1812   I'm going to close this and chalk it up to the "Mandela effect" for why I thought it was the AC being plugged causing the laptop to turn itself on.   I just flashed my C710 back to stock everything to test and it seems like it might have been the magnet that was making it appear as if it turned itself on.

closed 2018-06-11 by ShapeShifter499