MrChromebox/firmware #63: USB keyboard not functional

hawkshot2001, 2017-04-30:
The keyboard is a Gigabyte Force K83 on an Asus Chromebox M004U.  I updated to UEFI from SeaBIOS.  I can't type anything into the UEFI shell or select anything if I boot from a Manjaro USB or SteamOS USB (requires you to select boot from image).  This keyboard works fine with SeaBIOS and all but one of my other computers.  It works with a cheapo keyboard.

My Foxconn nT-A3500 has a similar issue if you use the latest firmware.  Using one version older is fine.

MattDevo, 2017-04-30:
unfortunately there's no good way to get diagnostic data from the UEFI portion of the firmware remotely, so not much I can do here without being able to reproduce the issue

hawkshot2001, 2017-04-30:
Is there a way I can get diagnostic data?

ReddestDream, 2017-04-30:
So, this keyboard doesn't work on one other machine? What processor generation/type does it have?

hawkshot2001, 2017-04-30:
It is an nT-A3500 nettop with an AMD E-350.

ReddestDream, 2017-04-30:
Interesting.

Does the keyboard light up at all when you plug it on the Chromebox or this nettop?

hawkshot2001, 2017-04-30:
The only lights are number lock, capslock, and a Windows key lock on the Chromebox.  The number lock and capslock do not light up at boot or if a button is pressed.  The Windows key lock is off at boot, but will toggle on and off.  The nettop behaved similarly before rolling back the firmware.

ReddestDream, 2017-04-30:
>The nettop behaved similarly before rolling back the firmware.

Do you have any idea what rolling back the firmware on the nettop did? That is, is there any changelog? I imagine not. But my first concern is that your keyboard is somehow not entirely compliant . . .

Does the keyboard work if you plug it in after an OS boots?

hawkshot2001, 2017-04-30:
Changelog for the nettop isn't much help all but the last two just say "Release new BIOS".  The latest one says "Support new HD Audio Codec chipset and combine the new & old HD Audio Codec Table".  The second latest is "Update new BIOS, support Brazos2 CPU".  I am running the third latest.

There isn't a USB logo anywhere on the device or the box, so it isn't certified.  I guess I figured it would be as it is name brand, but it is a cheaper mechanical keyboard so that must be where they cut costs.

I am installing an OS right now, so I will check to see if it works after boot.  After I comment on that, I'll start looking for a new keyboard and close.

ReddestDream, 2017-04-30:
Kk. Thx.

Two points:

1. If USB devices are out of spec, they can cause hardware damage and should not be used. If the keyboard is really bad/damaged/defective, it shouldn't be used.

2. There are some USB 2.0 devices that do not work correctly in USB 3.0 ports for whatever reason. This was more common when USB 3.0 was new and is fairly rare now.

hawkshot2001, 2017-04-30:
The keyboard works after OS boot.

The keyboard might be USB certified after all.  It looks like companies don't bother to put the USB logo on devices even though they pay a lot of money to USB-IF for testing.  I checked multiple devices at my house while the OS was installing and none of them have the logo on the device or user manual.  This was even the case for devices I know were certified.  USB-IF doesn't provide a list of what is certified either, so I don't even know if the Chromebox itself was certified.

I'm not ready to give up yet.  Is there some diagnostic data I can pull?

ReddestDream, 2017-04-30:
The problem must be how the firmware is detecting the device.

Unfortunately, that is very hard to debug remotely. We would really need the keyboard itself to be able to figure it out. :/

Here's something to try. Attach this problematic keyboard and a second working keyboard. On the working keyboard, press spacebar at the logo to enter the firmware menu. From there got to "Boot Maintenance Manager" and pick "Reset System." This will reboot the box. After this, see if the problematic works in the shell and elsewhere before OS load.

It might be that the firmware is not allowing enough time for the keyboard to be properly detected. This technique reboots without fully powering down USB devices, giving them more time to be properly detected.

If this technique works, the next thing to try is a powered hub between the keyboard and the box.

ReddestDream, 2017-07-16:
Planned to be re-examined for the next firmware release, along with other USB-related issues.

gdsports, 2018-08-12:
The problem may be the BIOS does not send SET PROTOCOL to set boot mode. The K83 in the default mode operates in NKRO mode which is very different from most keyboards. It works in boot mode if the SET PROTOCOL request is sent.

https://forum.pjrc.com/threads/53381-Gigabyte-keyboard-does-not-work-on-Teensy-3-6-host-port

ReddestDream, 2018-08-12:
It's an interesting idea, but I would hope it's not the issue. The UEFI USB driver that is part of TianoCore supports HID (more like how an OS does it) and should be robust enough to not cause issues. IIRC, I've also tested other USB NKRO keyboards and not had an issue with them in the UEFI.

MrChromebox, 2019-12-19:
I recently found/fixed a few keyboard init related issues in Tianocore, so if anyone here is still having problems drop me an email for a test build. Otherwise, fixes will be in the next release

MrChromebox, 2022-02-13:
@theflyingape there's absolutely nothing that can be done about that, you issue is with the stock Google firmware. Daskeyboard is known to operate outside of USB-HID spec, so not surprised it's having issues here

closed 2026-11-04 by MrChromebox