MrChromebox/firmware #65: Custom coreboot UEFI firmware for Toshiba CB2 2014 Swanky

jerrican62, 2017-05-13:
I want to replace ChromeOS on my Toshiba CB2 2014 Swanky.
Since I do not need the ChromeOS I decide to install MrChromebox UEFI full ROM and subsequently install  GalliumOS. Everything run out of the box and the performance is fast and I must give a lot of credits to MrChromebox and the GalliumOS team for their achievements sofar.
 I have met no issues however but one. And that is a serious showstopper.

I originally posted this in the GalliumOS support, but now identified it has to do with the combination of the UEFI ROM and my CB, and believe this is the right place to post this.
Booting from a full shutdown gives a starttime of about 15 minutes!
(all the time a black screen (backlit black for a fe seconds than black for a few seconds and then backlitblack until finally GalliumOS until finally GalliumOS appears).
Once booted into Galllium and using the restarting (insread of full shutdown) gives a restart in a acouple of seconds
I have already (re)installed/update the UEFI ROM, installed Linux Mint and reverted back to GalliumOS.
All the time the same issue, so the distro itself does not make any difference.

Waiting for 15 minutes to boot a laptop is not very inviting to use it, to say the least.
Are there other users having installed the UEFI Full ROM on a Toshiba CB2 2014 Swanky, or perhaps similar Celeron N2840 Bay Trail?

I would very much like to solve this issue because I am very enthusiastic about the GalliumOS on the CB.
I am a newby on Linux, just starting to use all of this a week ago, but read a lot of articles and with good documentation from different website so far am at this stage.

Any help or suggestions would be greatly appreciated.

ReddestDream, 2017-05-13:
Yeah. I've never seen this issue before.

AFAIK, @MattDevo has a Swanky too, and it doesn't have this issue on the UEFI firmware . . . :/

Best I think I can do is have you send the Chromebook to me.

I can externally reflash it and see if that resolves the issue, which it might If something really did go wrong during the firmware flash.

On the other hand tho, it might do nothing at all.

Have you tried reverting to stock and seeing if the issue now affects ChromeOS as well?

MattDevo, 2017-05-13:
The issue you are describing/experiencing is definitely not one we've seen before.  We've tested pretty much every Baytrail ChromeOS device, and hundreds of users have flashed the firmware without issue.  Additionally, my 2014 CB2/Swanky has no such issue. 

That's not to in any way imply that you did something wrong in any way, just that there's an issue there we haven't seen before.  I don't think remote debugging would be too useful, unless the firmware boot log showed something interesting, so your choices are really to send us the hardware for inspection, or revert to the BOOT_STUB firmware and just (re)install GalliumOS

jerrican62, 2017-05-13:
I will start at end:  I have got it working!

So what did I do:
Like suggested a reinstalled ChromeOS and everything worked, including fast boot after full shutdown. I then installed again Mr. Chromebox UEFI full ROM, put the GalliumOS as boot stick in the USB-port and once done immediately after reboot there was again this very long waiting time of about 15 minutes before the GalliumOS install screen appeared (I never had the " coreboot logo" and hence never choose to boot from USB stick but apparently it auromatically choose it). Nevertheless I installed GalliumOS but as I feared the issue reappeared. I therefore reinstalled ChromeOS and that works fine with the usual fast startup.

I do however have 2 Toshiba CB2 2014 Swanky chromebooks, and one CB is used as general chromebook by the family and I wanted to convert one into a linux laptop for personal use.
I therefore removed write-protection from the other one, acrivated developermode, installed MrChromebox UEFI FUll ROM and after reboot immediately got the rabbit-logo, pushed esc, and booted from the GalliumOS install USB-disc.  After installing Gallium both restarting and booting from full shutdown is fast and I always get the rabbit coreboot log for 1 second.
As you can imagine, I am a very happy guy!

I wondered what was the difference?
1) Either there is something wrong with the 1st CB, but then why does ChromeOS run flawelessly?
2) I used a slightly different installing procedure for the 2nd (succesfull) CB.
    For the 1st CB I followed the install instructions from "Fascinating Captain", " Install linux Mint on a Chromebook with a separate home drive" . I followed this for GalliumOS since at later stage I decided to use Gallium instead of Linux Mint (of course keeping in mind I wanted to fully replace the OS and not go into dual-boot, legacy or boot_stub)
for the 2nd CB I followed the installation instructions from the GalliumOS wiki.
The difference I found was that under the " installing/Preparing" instuction from Gallium OS under step 1: Enable Developermode there is an instruction to enable Legacy Mode booting with an instruction " sudo crossystem dev_boot_legacy=1"  at the shell, which I followed.
I do not know if that is relevant but it was not present in " Fascintaing Captains"  instructions.
The rest was similar.

With the GalliumOS on the 2nd CB installation and performance is perfect.

I have got what I want, one CB with ChromeOS for the family and one CB with galliumOS.

Thanks for your patience reading my questions and comments, and above all thanks for developing the UEFI full ROM, which is the best option to turn the CB into a real linux laptop, and I do not have to worry about draining the battery empty. Together with GalliumOS (also thank you deveopment team), this is the best opyion for me on the CB.

Thanks guys, I guess this issue van be closed.

closed 2017-05-13 by jerrican62