MrChromebox/firmware #67: My Chromebook HP 14 (Falco) is hotter when plugged, even with a fully charged battery

Paviluf, 2017-06-23:
Hello,

As [talked](https://www.reddit.com/r/chrultrabook/comments/6hmpby/my_chromebook_hp_14_falco_is_hotter_when_plugged/) with @ReddestDream the CPU of my Falco is 5°C hotter when plugged in, even if the battery is fully charged. When the Chromebook is fully charged, for the same fan speed, the temp should be the same, plugged or not.

I was wondering if there was some kind of "boost" when plugged, like a higher CPU clock or something but that don't seem to be the case. It's not that important and I'm not really concerned about it since the temps are perfectly fine but I open this issue just in case there is something to tweak in the firmware :)

I'm on Fedora 25 with full UEFI ROM and I use TLP.

Thanks !

ReddestDream, 2017-06-23:
Well, you confirmed before that the fans run at the same speed sequence plugged in or not and that is also confirmed by the code.

https://github.com/MattDevo/chrome-ec/commit/f6867fd923dbe8a85c945ab1ab7cb5fc43dfb769

I mentioned before that it might be due to the battery charging, which can put off heat.

You suggested that it might be due to some sort of boost. And I said here that that could be, but I haven't noticed it on Peppy, which has the same processor.

https://www.reddit.com/r/chrultrabook/comments/6hmpby/my_chromebook_hp_14_falco_is_hotter_when_plugged/djakl0h/

ReddestDream, 2017-06-23:
So, you are running Fedora. I don't know about TLP, but I have used Powertop and the settings there  vary plugged in vs. not. I'd check out TLP and see if it's changing with plugged in status.

Paviluf, 2017-06-23:
I just checked the cpu speed and plugged or not the speed is the same. 

Here is my tlp settings. I think it's the default one.
[tlp.txt](https://github.com/MattDevo/firmware/files/1097989/tlp.txt)

ReddestDream, 2017-06-23:
Just looking at this, Turbo boost is different on battery vs. plugged in and so is P-state performance. Several other values are different too.

ReddestDream, 2017-06-23:
# Minimize number of used CPU cores/hyper-threads under light load conditions
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1

This might be it. It's CPU temps that are the issue, right?

Paviluf, 2017-06-24:
I made some tests. They are not perfect since I didn't always shutdown the computer and wait for it to cool down between each tests (but I let it idle for about 10mn to report the temp) but it give us a rough idea of the effect of tlp.

**Fedora 25 (Gnome)**
On battery with tlp (my settings): 41°C
On ac with tlp (my settings): 46°C

On battery without tlp: 46°C
On ac without tlp: 48°C

_Here I shut down the computer_
On battery with tlp (same tlp settings for battery and ac): 40°C
On ac with tlp (same tlp settings for battery and ac): 42°C

**Opensuse Tumbleweed (KDE)** (without any tweak - clean install)
On battery without tlp: 51°C
On ac without tlp: 54°C

It seem that tlp was in part responsible, so thank you, I adjusted my tlp settings. There is still a 2 to 3°C increase in temp but the battery was not fully charged when I tested.

ReddestDream, 2017-06-24:
>It seem that tlp was in part responsible, so thank you, I adjusted my tlp settings.

👍

>There is still a 2 to 3°C increase in temp but the battery was not fully charged when I tested.

That is within the margin of error and/or temperature stabilization and/or the effect of charging.

Paviluf, 2017-06-25:
I tested with the battery fully charged and my adjusted tlp settings. On battery the temp was 41°C and on AC the temp was 42°C so it's great. Thanks !

closed 2017-06-25 by Paviluf