MrChromebox/firmware #68: Haswell UEFI : CPU idle states problem

mpoly, 2017-07-02:
Hi there,

On both my Peppy and Panther with UEFI firmware the CPU have idle states issues, resulting in high power consumption.
Idle states stats with UEFI FW : https://i.imgur.com/PoofWIu.png
Idle states stats with johnlewis.ie LegacyBIOS : https://i.imgur.com/fdksvW0.png

It isn't very problematic on Panther because it isn't a laptop, but on Peppy it's catastrophic, the battery time is divided by two!

The UEFI FW on Peppy has others issues (but not on Panther) : the track-pad randomly freeze, sometimes the whole PC. I had to switch back to johnlewis.ie LegacyBIOS until this is fixed.

ReddestDream, 2017-07-02:
>The UEFI FW on Peppy has more issues (but not on Panther) : the track-pad randomly freeze, sometimes the whole PC.

So, there is a rare bug in the Linux driver for Cypress touchpads that causes them to become unresponsive. The bug is random and generally requires a reboot to fix once it happens. This bug has been around for a long time and has nothing to do with the firmware. If you have dmesg from around the time that happened, we can check for the bug.

>On both my Peppy and Panther (Haswell) with UEFI firmware the CPU have idle states issues, resulting in high power consumption (tested on Linux kernel 4.4, 4.8 and 4.10).

So, from what I can tell, in your UEFI FW screenshot, the CPU is not actually idle. That is, there is likely background activity you are not aware of that is preventing the CPU from entering higher C states. Please confirm. I've not noticed any issue with idle CPU states on Haswell.

mpoly, 2017-07-02:
But back to johnlewis.ie FW, both problems are gone.

About CPU idle, my testing was : I power on Peppy, run powertop, and wait 10 minutes. I did both tests with the same linux distro (Linux Mint 18.1 which use Ubuntu kernels). I did it because I noticed the battery time was a lot shorter. I did it multiple times, always the same result. Powertop report 8W discharge on UEFI FW but 5W on LegacyBIOS.

About touch-pad freeze, I used johnlewis.ie FW during years without any freeze. When I switched to UEFI FW, it suddenly happened a lot. Back to LegacyBIOS, it looks gone again.

ReddestDream, 2017-07-02:
>[EDIT] I'm flashing UEFI and reinstalling on another SSD to try another time.

Thx.

mpoly, 2017-07-02:
Then, I flashed UEFI, did a fresh Linux Mint 18.1 install on another SSD and I confirm the problem.
More screen-shots : http://imgur.com/a/miS7i

ReddestDream, 2017-07-02:
So, I have a few questions:

1. Are you installing the same distro (Linux Mint 18.1) for both your UEFI and JL tests? You won't be able to use the same installation itself w/o some tricks with rEFInd.

2. Can you confirm with top or whatever you'd like to use that there are no differences in the running processes between your UEFI and your JL tests?

If there really is a problem with the CPU powerstates, it would have to be an issue with Coreboot itself, which both ROMs use, rather than the UEFI (TianoCore) vs Legacy BIOS (SeaBIOS) payload. The UEFI one simply uses a newer Coreboot codebase.

mpoly, 2017-07-03:
I haven't started my "tests" today, then some history : I used JL.ie LegacyBIOS on Peppy during years with elementary OS on stock SSD. Everything worked fine. Then one month ago, I did lots of changes : Switched to mrchromebox.tech UEFI (to use GPT partitioning), changed the SSD for a bigger one (64Go) and installed Linux Mint 18.1. I did it on my Panther too (except I kept the stock SSD) as it had an old mrchromebook LegacyBIOS. Then to do that I made an USB pen with Linux Mint 18.1 installation on it, I use this one for every installations now, it boot on both firmwares.

Then I ran this new setup during about 3 weeks. During that time I noticed on Peppy frequent touch-pad freezes that I never had before and shorter battery time (by a lot, hours). Then I started to "play" with Intel PowerTOP to tune power management. I let SATA power-save OFF because the new SSD is a Transcend and has bugs in "low_usage". I set the backlight to 80%. Others power-saving stuffs are ON (I tried OFF too for any touch-pad related devices, but still buggy). While playing with PowerTOP, I noticed Core0 is never idle.

Then as it wasn't stable, I had 3 hypothesis : UEFI firmware problem, SSD problem or Mint problem. As the firmware is THE main component, I started with this and flashed back to JL.ie and reinstalled Mint (because as you said, same install can't boot on both firmwares). Then I'm using this setup since a week and everything became stable again, no more touch-pad freeze (but I need to use it one more week to be very sure) and way longer battery life. Then I ran PowerTOP, and saw everything is now fine with Core0 idle and it report 5W consumption instead of 8W (I really gain some hours of battery).

I ran PowerTOP on Panther UEFI too, and I saw similar stats than Peppy UEFI.
Then yesterday I opened a thread on github to let you know. As today you answered, I decided to try UEFI fw again, then I flashed it. But as it can’t boot anymore the installed Mint, I removed the new SSD too and put the stock one. Then I installed Mint on the sock SSD and set backlight to 80%. And I noticed the same things as before in PowerTOP (8W usage, Core0 never idle). Then I did screenshot, flashed JL.ie again, swapped to new SSD, boot, backlight to 80%, ran PowerTOP again, screenshots because it’s back to 5W and Core 0 idle was fine and then posted on github with comparison.

And of history, then :
 
1. Yes, both tests are running on Linux Mint 18.1 with Ubuntu kernel 4.8.0-58. Same backlight setting, tested UEFI fw with Transcend and stock SSD.

2. I use exactly the same USB pen to start installation on both firmwares.

3. I’ll flash UEFI fw again to make htop screen-shots, but I noticed nothing special when I looked into it some times ago. And today’s Mint installation was fresh (on stock SSD to not broke the GPT/UEFI install on Transcend SSD).

4. I forgot to turn off Bluetooth in latest UEFI screenshot, but it change nothing, I did it in the past. Both screen-shot have everything turned ON except SATA, on both firmwares.

5, As today I ran UEFI fw only one hour, I can’t confirm touch-pad freeze. I will keep JL.ie firmware one more week to be sure there is no issues, even on Linux Mint. And if no freeze, I’ll reinstall UEFI fw for some days to see if it freeze again.

ReddestDream, 2017-07-03:
Couple things:

Yes, make sure Bluetooth is off. Turn SATA power-save on on UEFI. Other people have said that current Transcend drives do not have the bug while on the UEFI firmware.

Thanks for your answers to my questions above. One more follow up question:

1. Are you testing CPU usage on battery or plugged in? This will make a difference to PowerTop and the power states the CPU enters.

mpoly, 2017-07-03:
I already tested with Bluetooth off and everything, no improvement. The only one thing I haven't tested yet is to try another Linux distro, not based on Ubuntu.

1. I test it on battery, otherwise battery discharge isn't displayed.

I can confirm this bug too : https://github.com/MattDevo/firmware/issues/52. Sometimes at boot the screen stay black, and have to power off and power on again.

ReddestDream, 2017-07-03:
Did you try to enable SATA power-save/Link Power Management?

mpoly, 2017-07-03:
Yes, on my Panther SATA power-save is enabled, and I have the same "idle" problem.

ReddestDream, 2017-07-03:
Oh. I guess John updated his ROMs. Well, his is newer than ours now and likely has this CPU idle bug fixed. Ours should be fixed the next time we issue an update.

mpoly, 2017-07-03:
Ok, I'll keep an eye on your homepage to see when you'll update it, then I'll update Panther first.

ReddestDream, 2017-07-04:
So, you'll be happy to know that I have reproduce some of your results on a newer UEFI ROM. :)

Now, w/o the contrast to the JL ROM, I would say that there is nothing wrong and assume that there is simply enough background activity to prevent 1 Core (usually Core0 both because it is numbered first and sometime the "best" core on the die is in fact numbered Core0) from completely entering the "Idle" state. I still don't completely understand what "Idle" in Powertop means since the lowest possible clock speed for both cores on this processor is ~800 MHz (8.0 * 100 MHz FSB). But it seems like those clock speeds aren't the point since they are similar on both the UEFI and JL ROMs.

At one point, I was able to observe the two Cores essentially "trading places," with Core0 mostly in "Idle" and Core1 not, and I was able to get discharge rates under 8 W, but nowhere near the 5 W seen on JL's ROM in your example.

I'm going to flash the JL ROM and see if I get results similar to yours there.

ReddestDream, 2017-07-04:
So, one thought I have is that "Idle" represents how much time the CPU has spent in a particular state recently and "Idle" just represents the percentage of CPU time/cycles not spent on anything at all. What bothers me is still the wattage tho and the ease with which both Cores seem to idle on the JL ROM.

mpoly, 2017-07-04:
I think the "Idle stats" tab is more informative as the problem looks to be in c-states, not in p-states.

I'm doing more researches too (on Panther because it's easier for me), then I disabled intel_idle in kernel boot parameters to force the use of acpi_idle instead, but "C0 active" is still too high on Core 0. I ran htop as the same time, no consuming process is present : https://i.imgur.com/3d2SLin.png

ReddestDream, 2017-07-04:
>I think the "Idle stats" tab is more informative as the problem looks to be in c-states, not in p-states.

See, from my tests in Linux and Windows, there are no missing or disabled C-state or P-States. Both Cores can enter all states.

>So, one thought I have is that "Idle" represents how much time the CPU has spent in a particular state recently and "Idle" just represents the percentage of CPU time/cycles not spent on anything at all. If that's the case, the term "Idle" is meaningless and all that matters is the Frequency, which is what I thought from the beginning.

I still stand by this. I think the "Idle" percentage refers to how much of the CPU is idle, not a C-state or P-state metric. It is related to C-states and P-states tho in that the CPU cannot/will not drop to lower frequencies and higher C-states when there is too much activity. So, in order for those changes to occur, a high "Idle" percentage is required.

mpoly, 2017-07-04:
Hum yes, you have a 100% idle on Core0 on 2nd screen-shot, why not me?? :p It never happen for me...

ReddestDream, 2017-07-04:
Well, it didn't last very long. ;P

Part of me trying to figure out what "Idle" really meant.

I'm not denying there's an issue. I'm just trying to figure out exactly where it is.

mpoly, 2017-07-04:
I understand, no problem.

Well, P-states and C-states are very different, P-sates are the global frequency of the core, while C-states disable only some subparts of the core.

ReddestDream, 2017-07-04:
Well, C-states are also very sensitive to things outside of the CPU.

For example:

"If the CPU/SoC and much of the motherboard can aggressively drive down to S3-like idle power, the display becomes an even bigger burden than it has in the past. To allow the entire platform to go to sleep while the display is still active, Intel recommends Panel Self Refresh (PSR) to be implemented in Haswell Ultrabooks. The idea behind PSR is to put DRAM on the panel itself to store the frame buffer. In the event of a static display (e.g. staring at a Word document, looking at your desktop), the GPU frame buffer is copied to the panel’s DRAM, and the GPU itself is shut off instead of having to drive meaningless content updates to the display 60 times per second. With PSR, the SoC can go into its deepest sleep state (C10)."

http://www.anandtech.com/show/7047/the-haswell-ultrabook-review-core-i74500u-tested/3

mpoly, 2017-07-04:
> If we can say that the frequencies (P-states) and Idle percentages are /not/ the issue, we might be looking at something that isn't actually the CPU at all. Might be something else not powering down properly and indirectly preventing the CPU from powering down.

Yes, I think something is preventing the CPU to reach higher sates, but they are present and usable.

ReddestDream, 2017-07-04:
My first thought is the display. The UEFI firmware uses a different method of initializing the display than the Legacy SeaBIOS one. John is able to just use the stock VBIOS while we have to transfer those settings as best as possible into something to use with GOP. In addition, in order to support the Intel driver on Windows, we had to break ACPI backlight controls and just route them through the Intel driver. The Linux Intel driver seems fine with this since we put something in to signal it /not/ to use the broken ACPI backlight controls and to fallback on the driver. Either of those might somehow be preventing the Integrated GPU from powering down fully. Now I'm not sure since the GPU seems to be in RC6 mostly on both JL and UEFI ROMs . . . Could be something else entirely.

mpoly, 2017-07-04:
> One thing I want to rule out is the display. If you are willing, the UEFI firmware boots with no display connected. Just disconnect the battery before you unplug it. Then SSH in and see if anything changed.

I can't now but I'll try tomorrow. Probably on Panther because it will be easier to just unplug HDMI. But firt I'll have to reinstall Mint because currently there is dm-crypt at boot then it will be stuck asking the password and won't run SSH server.

mpoly, 2017-07-04:
Then I did it on Panther : Unplugged the screen, boot, connected to SSH via ethernet, ran PowerTOP, but I saw absolutely no improvement.

I can't try on Peppy as it's my main laptop then I don't want to break anything, and it has no ethernet port.

Paviluf, 2017-07-07:
Just to let you know that I see no problem on my side with my Falco that has a haswell too. The drain rate is only about 4W with Fedora 25 with TLP and minimum brightness. 

[image]

ReddestDream, 2017-07-16:
New firmware (07/14) is out if you would like to test on that.

mpoly, 2017-07-16:
Nice :) I'm going to test it on Panther first.

ReddestDream, 2017-07-21:
@Paviluf Are you around? Sorry this is off topic, but I'm hoping you can confirm something with Falco's mic on the new firmware.

Paviluf, 2017-07-29:
@mpoly maybe you could try on fedora with tlp to check if the problem occur too ?

mpoly, 2017-09-03:
@Paviluf Hello, I was on vacation. Yes, I'll try it next week.

Paviluf, 2017-09-14:
Well I can confirm the problem with Kubuntu 17.04.

ReddestDream, 2017-09-14:
Yeah. I'm not sure what's causing this or how much of an issue it really is . . .

Paviluf, 2017-09-14:
Well it divide the battery life by 2, so that's a big issue :D

ReddestDream, 2017-09-14:
>Well it divide the battery life by 2, so that's a big issue :D

I'm not convinced of that. ;p

Paviluf, 2017-09-14:
I didn't time the battery life on Ubuntu but on Fedora I get something like 8 hours in really usage. Powertop report something like 12h with lowest brightness on Fedora. On Ubuntu powertop report something like 7h.

Powertop is reliable tool it seem.

mpoly, 2017-09-15:
> Well it divide the battery life by 2, so that's a big issue :D

> I'm not convinced of that. ;p

There is a real battery loss, that's why I installed PowerTOP to find what's wrong... And opened this thread.
If it's better on Fedora, that's probably a setting in kernel compilation, related to UEFI.

Paviluf, 2017-09-16:
I tried John Lewis firmware and the battery life is the same on Fedora and Ubuntu (about 11h reported by powertop) !

MattDevo, 2017-09-17:
I'm not able to reproduce this here under equal conditions.  I tested JL's legacy firmware, as well as UEFI and Legacy firmware built from my tree on Wolf (Dell CB11 2013 / HSW 2955U).  On Fedora 26 Live USB (kernel 4.11.8), and with all tunables in powertop set to 'good,' all 3 firmwares show similar power consumption/battery discharge rate (around 5W w/normal brightness) and show both CPU cores spending most of their time idling at C10.

Paviluf, 2017-09-18:
Yeah, that's not the problem :) The battery life with Fedora (25, 26) is great (< 5w) with all the firmware. It's with Ubuntu (and Opensuse at least) that the battery life is bad ( about 8w).

MattDevo, 2017-09-18:
so how is this a firmware issue then?

Paviluf, 2017-09-18:
Well it work well with JL firmware.

MattDevo, 2017-09-18:
> Well it work well with JL firmware.

it worked under legacy boot vs UEFI boot; the fact that there's no issue under Fedora with same kernel version strongly suggests it's a kernel issue.  If it were a firmware issue preventing the CPU from going into low power idle states, it would very likely be present across all kernels tested

Paviluf, 2017-09-18:
It seem logical but it's unlikely that only Fedora managed to get good kernel settings for uefi.

ReddestDream, 2017-09-18:
>It seem logical but it's unlikely that only Fedora managed to get good kernel settings for uefi.

Doesn't sound too far-fetched to me, actually. A lot of machines still use Legacy BIOS and, even on newer machines, a lot of Linux users just use the CSM since they are told it "works better for Linux," probably because it ensures that Secure Boot is disabled.

mpoly, 2017-09-18:
The problem occur at least on Peppy and Panther with Ubuntu and derivatives (Linux Mint in my case).
On my Intel UEFI barebone with same OS, the CPU enter every idle state. 
That's very strange I know...

MattDevo, 2017-09-18:
Debian 9.1 behaves properly on Wolf w/UEFI, so the issue seems to be isolated to Ubuntu and derivatives (Mint, GalliumOS)

Paviluf, 2017-09-18:
Opensuse have a the problem too.

jfrankenau, 2018-02-12:
I'm experiencing the same issue using an Acer C720 (Peppy) with NixOS 17.09. Only with John Lewis's firmware the CPU enters idle while the battery discharge rate is below 4W compared to around 6W for the UEFI firmware. I've been running both tests with identical configurations.

ReddestDream, 2018-02-12:
Yeah. This is a known issue on the UEFI ROM. Not sure how to resolve it since AFAICT, it's an issue with TianoCore itself on HSW+ just not being fully optimized . . .

MattDevo, 2018-02-14:
so we've narrowed this down to a coreboot (not Tianocore) bug which was introduced sometime between 2014-08-01 and 2016-09-06.  we're working on identifying the problematic commit/issue but will take some time

mauritsvdvijgh, 2019-04-30:
I am experiencing this same problem using arch linux on a peppy/C720P chromebook. 
Is there any chance that this will get fixed in the firmware or is there a workaround/setting that can be used for the kernel/os? Would really like the cpu to be able to go to sleep and extend the battery life.

Maybe not that helpful seeing as you have found the problem already:

This is some data I collected on 2 different kernels. If not specified the data is gathered running on battery. Changing the last few settings powertop denotes as "Bad" to the "Good" setting did not make a difference.
Willing to gather more data/try other things if it helps. 
[debug.zip](https://github.com/MrChromebox/firmware/files/3131809/debug.zip)

MrChromebox, 2019-04-30:
@mauritsvdvijgh as you noticed, it's not an issue with the older 4.9 kernel, so is it a firmware issue or a kernel one?  I've simply not had the time necessary to diagnose the issue to determine root cause. And I don't know when I might unfortunately

mauritsvdvijgh, 2019-04-30:
Not going into C states happens in the same way on both kernels and with all settings/kernels I have tried so far.

Only difference is 4.9 not showing high hardware interrupts: https://github.com/MrChromebox/firmware/issues/110#issuecomment-487985768

>  I've simply not had the time necessary to diagnose the issue to determine root cause. And I don't know when I might unfortunately
I understand. 
Thanks for all the effort put into this firmware and the firmware installer tool!

MrChromebox, 2019-04-30:
ok, thanks for clarification

MrChromebox, 2020-03-18:
fixed in 2020.03.17 release!

closed 2020-03-18 by MrChromebox

mpoly, 2020-03-18:
Really? I'll give it a try then. \o/