MrChromebox/firmware #69: Fan on Panther does not run under uefi install of Windows 10

MercuryMan2900, 2017-07-04:
Thus, my poor Panther runs really HOT.  I do not have this problem under Linux.  Debian 9 for example, runs flawlessly.

MattDevo, 2017-07-04:
it runs, Windows just keeps it in the lowest/slowest state for some reason.  If you install a utility like Speedfan and set it to run at startup, then you won't have any issues.

closed 2017-07-04 by MattDevo

MercuryMan2900, 2017-07-04:
Oh!  Thank you SO Much!  It solves the problem!

Sent from Windows Mail

From: MrChromebox
Sent: ‎Tuesday‎, ‎July‎ ‎4‎, ‎2017 ‎11‎:‎41‎ ‎AM
To: MattDevo/firmware
Cc: MercuryMan2900, Author

it runs, Windows just keeps it in the lowest/slowest state for some reason. If you install a utility like Speedfan and set it to run at startup, then you won't have any issues.