MrChromebox/firmware #70: Current legacy full ROM on Zako is broken; UEFI full ROM does not load XenServer

igel-itzo, 2017-07-10:
Sending this in not only to see if it can be fixed, but to save some poor soul from bashing their head against the wall. 

I have a zako (HP) box, that had a legacy boot full rom from 2015 (did not remember the month&day). It was successfully running xenserver (6.5 and then 7.1). and also openelec/libreelec from a SD card prepared with grub4dos. 

I decided to upgrade to the latest legacy full rom, but it was not working - both xen and libreelec would not start - as soon as the kernel loaded, the screen would go blank, and there will be no network activity at all (verified on my DHCP server, no attempt to get an address were made). Lubuntu live did load though. 

I attempted to install UEFI full rom, but that was completely unable to locate the bootloader, even when going through the tianocore menus. The same USB disk with Xen booted right away in a HP Steam mini with UEFI boot. 

The only way I was able to get XenServer working again - restore stock rom (I had a backup), update RW_LEGACY, update GBB flags.   

It would be really nice if the full legacy rom is fixed or the full UEFI rom is capable of booting XenServer (even if with manual adding of the bootloader). 

P.S. at some point I had to restore ChromeOS - an old recovery image from late 2014 did not work - it would restore but the boot loader will still insist chromeos is damaged. Had to create a new image. The new image has "enable debug options" screen during the start - don't touch that, it will set a password on the root in chromeos, and it won't be the one you put in.

ReddestDream, 2017-07-10:
>It would be really nice if the full legacy rom is fixed or the full UEFI rom is capable of booting XenServer (even if with manual adding of the bootloader).

This is possible to do with the rEFInd bootloader, tho it takes a bit of setup:

https://sourceforge.net/projects/refind/

rEFInd can boot Linux kernels directly on a UEFI firmware even if the OS installation is done in a Legacy BIOS style. This is not advised as a permanent solution, however, and reinstallation in UEFI style is strongly recommended. I can help you get this set up if you would like.

BTW, our UEFI ROM is lacking two features: 

1. CSM (Legacy Boot) Support: This is not planned to be added as all modern, fully-developed and supported OSes in 2017 are capable of booting with UEFI. Old Legacy Linux-based OS installations can booted with rEFInd without the need for a CSM, tho this is not a permanent solution and reinstallation is advised.

2. NVRAM Support: This is needed so that that the UEFI firmware can save its settings and use non-default bootloader paths. Unfortunately, the open-source codebase we use does not have NVRAM capability in the Coreboot-compatible UEFI package. We are hoping this may be added as time goes on, but it would require extensive work to add this new capability. :/

>It would be really nice if the full legacy rom is fixed or the full UEFI rom is capable of booting XenServer (even if with manual adding of the bootloader).

Our Legacy Full ROMs are considered deprecated and development efforts are focused on the UEFI ROMs. John Lewis is continuing work on Legacy Full ROMs. His latest is from 05/22/2017. I'm not involved in John's work, however, so I do not know what issues you might encounter on his ROMs.

https://johnlewis.ie/custom-chromebook-firmware/rom-download/

igel-itzo, 2017-07-10:
Hi ReddestDream. 
I am familiar with rEFInd, I use it quite a bit on my macs. Tianocore was not finding any EFI partition at all on my USB drive even though it did find the drive itself, so I am not sure rEFInd would help (I have removed the M.2 drive), as it will likely not be able to find rEFInd itself in the first place. I'll give it a try some day when I have time to tweak firmware again, hopefully next weekened and report.

As I said, part of the purpose of my post if also to provide some pointers to others who end up in the same situation as I did, and how to get out of it. Stock + updated RW_LEGACY + GBB flags (to do legacy boot after 1s delay) works correctly.

MattDevo, 2017-07-10:
> Stock + updated RW_LEGACY + GBB flags (to do legacy boot after 1s delay) works correctly.

There's no issue with the current Legacy Full ROM for Panther/Zako/Tricky/McCloud, I've tested it here and it boots all Legacy OSes without issue.  I'm not sure what the problems you ran into are, but I don't think they are firmware-related.

closed 2017-07-10 by MattDevo