MrChromebox/firmware #72: me_cleaner (on acer c910-354Y)

Popkultur, 2017-08-13:
Hello,

I really like your project so far! I am running an acer c910-354y chromebook (Broadwell) on your full custom UEFI and I'd like to know if someone already managed to wipe the IME from the bios (especially with this model). As a former Windows user, I'm just diving into Linux and openBSD stuff, and I think I am not yet able to reproduce the neccessary steps myself. So if there is a possibility to have some support with flashing and especially rebuilding coreboot without IME I'd appreciate it. I wanted to use me_cleaner, however it seems I do not know how to rebuild coreboot and if it is possible after all. I don't mind a brick, however I only have one try so far. What would be needed for this to work out? Would it be sufficient to provide a backup bios image created with flashrom? Thanks a lot!

kind regards

MattDevo, 2017-08-13:
I can easily build firmware images with ME Cleaner applied, the problem is that they must be flashed externally (using a RPI, eg) since the ME region of the flash chip is not writable with the system powered-on/booted.  Likewise, you can read the current firmware with a RPI, run ME cleaner on that, and then write it back with the RPI.  There's no way around needing an external programmer to do this, which is why the images I distribute don't touch the ME.

Popkultur, 2017-08-13:
I just read about internal flashing at corna/me_cleaner where it states it should be possible, so I will try to build ifdtool and see if I get a good response, and if that works I will just try to flash it internal. I removed the write protection screw so it might work.

MattDevo, 2017-08-13:
@Popkultur you absolutely cannot flash it internally via flashrom -- the IFD is locked to prevent that.  If the IFD were unlocked you could, but it is not.

Popkultur, 2017-08-13:
I will try to flash it externally then with a RPI. Thank you.

Popkultur, 2017-08-18:
I have tried to flash the Winbond W25Q64FV Chip by SPI with a raspberry pie 1 Model B. However, I can not get it to work, it does not recognize the chip. Either the connection ist not correct, or there is some other issue. However, it says calibration success, so I believe it has a connection with the chip? I have tried several settings including power by raspberry as well as power by board. I also tried with and without pin 3+7 powered. I believe there might be a problem with my rasberry pie and the OS and some drivers. I connected PINs as follows:

1 (CS) > 24
2 (DO) > 21
4 (GND) > 25
5 (DI) > 19
6 (CLK) > 23
[ 3,7,8 (VCC) > 17 ]

I tried compiling the flashrom from source, however it fails. It tells me some constant is defined but not requested. I need to log the exact error. If I try to compile with keep-going, it fails to write "all". If using the rasbpian apt-get flashrom, it won't work as described above.

I seriously don't have a clue where to look. Maybe I need to check if I have any output or Voltage at my raspberry at all.

MattDevo, 2017-08-18:
@Popkultur don't connect pins 3/7, only 8 needs vcc.  Use this writeup and see if that helps you with compiling flashrom: 
http://sicarul.com/how-to-un-brick-your-toshiba-chromebook-2-gandof-without-invoking-any-demons/

you can test identification of the flash chip with sudo flashrom -v

MattDevo, 2017-10-20:
closing this since it's not an actual issue

closed 2017-10-20 by MattDevo