MrChromebox/firmware #73: Unintended suspend

Paviluf, 2017-09-11:
Hello,

My HP Chromebook 14 (Falco) sometime go to sleep without any reason on Fedra 26. I ran F25 for month without any problem. By the way I briefly tried F26 KDE and often the laptop couldn't be suspended (hang on a black screen).

Do you have an idea of what can produce that ? I also opened a bug report against Fedora :
https://bugzilla.redhat.com/show_bug.cgi?id=1490516

Thanks !

Paviluf, 2017-09-12:
It seem the there is fake "Lid closed" event...

`sept. 12 19:59:30 hp systemd-logind[759]: Lid opened.
sept. 12 19:59:30 hp systemd-logind[759]: Suspending...
sept. 12 19:59:30 hp systemd-logind[759]: Lid closed.
`

ReddestDream, 2017-09-12:
Are you on the latest firmware (07/14/2017)?

Paviluf, 2017-09-13:
Yes since July.

Paviluf, 2017-09-13:
Well my bug on Fedora have been closed and they said that is surely a firmware problem. What's surprise me is that I didn't had any problem with Fedora 25 but they are probably right since a lid event seem to be triggered.

ReddestDream, 2017-09-14:
Yeah. I can't guarantee it's not a firmware issue. Can't blame them for closing it. But it didn't happen on Peppy during my firmware tests in Mint and the hardware is almost identical to Falco.

Paviluf, 2017-09-14:
I will check with an other distro. Thanks !

MattDevo, 2017-09-17:
does this issue occur with the default 4.11.8 kernel, or just the 4.12 one were running in the other issue post?  Is it going to sleep while you're using it, or just shorter than the preset timeout?

Paviluf, 2017-09-17:
I think it happened with kernel 4.12. It went to sleep while I was using it.

MattDevo, 2017-09-17:
sounds like Fedora's 4.12 kernel is a big hot mess

Paviluf, 2017-09-18:
No issue with JL firmware for the last 2 days.

Paviluf, 2017-09-21:
Well I had an unintended suspend with John Lewis firmware. Either it's a Fedora 26 problem or my lid switch is broken.

closed 2017-09-21 by Paviluf

Paviluf, 2017-09-21:
Do you know how can I test my lid ? @MattDevo What is the preset timeout you talked about ?

MattDevo, 2017-09-21:
> Do you know how can I test my lid ? 

test in what sense?  Perhaps you can check the EC console log (using ectool) to see when the open/close events are being triggered, but not sure exactly.

> What is the preset timeout you talked about ?  

I was referring to whatever power management settings are such that the device display turns off or goes into suspend after a certain idle time.

Paviluf, 2017-09-21:
I meant a way to see if the open / close event are really triggered by the lid.

MattDevo, 2017-09-21:
that's what the EC console log would tell you - if the EC detected a change in lid status, and triggered an ACPI event for the OS to handle

Paviluf, 2017-09-22:
Ok I will try. It only happened once in 5 days with JL firmware so it won't be easy to find out.

By the way, on the first post, I mentioned that I briefly tried F26 KDE and often the laptop couldn't be suspended (hang on a black screen). With JL firmware it didn't happen once.

Paviluf, 2017-09-25:
> I briefly tried F26 KDE and often the laptop couldn't be suspended (hang on a black screen). With JL firmware it didn't happen once.

Well it happened with JL firmware...

> sounds like Fedora's 4.12 kernel is a big hot mess

You seem to be right !