MrChromebox/firmware #74: Error messages on boot

Paviluf, 2017-09-13:
Hello,

On Fedora 26 I have these message on boot with Linux 4.12. The usb one is harmless (https://github.com/MattDevo/firmware/issues/55) but what about the rest ? Is the firmware or the kernel responsible for these messages ?

```
sept. 12 00:56:17 hp kernel: Couldn't get size: 0x800000000000000e
sept. 12 00:56:17 hp kernel: MODSIGN: Couldn't get UEFI db list
sept. 12 00:56:17 hp kernel: Couldn't get size: 0x800000000000000e
sept. 12 00:56:18 hp kernel: usb 1-4: string descriptor 0 read error: -22
sept. 12 00:56:19 hp kernel: [drm:intel_set_cpu_fifo_underrun_reporting [i915]] *ERROR* uncleared fifo underrun on pipe A
sept. 12 00:56:19 hp kernel: [drm:intel_cpu_fifo_underrun_irq_handler [i915]] *ERROR* CPU pipe A FIFO underrun
```

Thanks !

ReddestDream, 2017-09-14:
I don't test 4.12 yet, so I can't guarantee compatibility. I only tested 4.10 for this release.

The next firmware release will be tested on newer kernels.

Paviluf, 2017-09-14:
Yes, it's a new message. It happen with kernel 4.13.1 from repos.fedorapeople.org/repos/thl/kernel-vanilla.repo too. Doesn't happen with kernel 4.11.8 that I will use for now.

Fedora bug report https://bugzilla.redhat.com/show_bug.cgi?id=1491521

Thanks !

Paviluf, 2017-09-16:
I tried John Lewis firmware and I don't have the messages !

MattDevo, 2017-09-17:
the usb 1-4 error is an issue reading the descriptor for the Atheros BT device and can safely be ignored.

the MODSIGN error seems to be related to secure boot - see: 
https://forums.fedoraforum.org/showthread.php?t=315207

the fifo underun error is another driver/kernel bug, see:
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1550779

as both of these issues are related to UEFI/UEFI graphics init, it makes sense that you wouldn't see them on John Lewis' Legacy firmware.  And as they're not present on the mainline 4.12 kernel (or earlier Fedora kernels even), it sure seems like a Fedora/Fedora config issue than one with our UEFI firmware.

closed 2017-09-17 by MattDevo

Paviluf, 2017-09-18:
Ok thanks !

Paviluf, 2017-09-21:
Hello,

Can you provide the infos needed on the Kernel bug tracker please ?
https://bugs.freedesktop.org/show_bug.cgi?id=102759

I would have do it but I can't reinstall the firmware for now since it's needed to reinstall Linux.

Thanks !

Paviluf, 2017-10-12:
I briefly tried Ubuntu 17.10 (kernel 4.13) and the messages were present.