MrChromebox/firmware #75: The firmware seem to have some problems

Paviluf, 2017-09-16:
I tried John Lewis firmware because I have some problem after installing Fedora 26 and I was surprised to saw it is working much better. There is no error message on boot https://github.com/MattDevo/firmware/issues/74 and the battery life is great and similar across distro https://github.com/MattDevo/firmware/issues/68. Also I thought that Ubuntu wasn't working great on my Falco because it had some stability problems but it seem to work perfectly well with John Lewis firmware !

I hope these quirks can be fixed. Thanks !

MattDevo, 2017-09-16:
I'm not sure why you opened another issue ticket just to reference two existing ones.  Closing this as there's no addressable issue here

closed 2017-09-16 by MattDevo

Paviluf, 2017-09-17:
Sorry I should have be more precise. I opened an other issue because it seem that Ubuntu have some stability problems with your firmware but it seem to work perfectly well with John Lewis one. Before trying JL firmware I thought that was Ubuntu that was buggy. But since it doesn't worked well I didn't test it that much.

Paviluf, 2017-09-17:
I think part (all ?) of the problem that I mention is related to https://github.com/MattDevo/firmware/issues/52 because I often had a hard time to boot ubuntu when I wanted to test it.