MrChromebox/firmware #76: SeaBIOS does not recognize large SD card (128G)

nebulakl, 2017-10-16:
Hi I'm using your RW_LEGACY firmware on CAVE and it doesn't recognize the 128G SD card when I tried to boot it using my laptop's build-in card reader. Booting the same SD card with a usb card reader works fine.

MattDevo, 2017-10-17:
define not recognized?  as in, not listed in the boot menu when you press ESC?  Or listed but fails to boot?

nebulakl, 2017-10-17:
it's not in the boot menu

MattDevo, 2017-10-17:
looks like Cave uses a PCI-connected SD card reader, as opposed to an internal USB one like Chell (eg).  I'm not sure SeaBIOS supports such devices but I'll have a look and see.  If you're able to provide me with a firmware boot log, that would be quite helpful.  email me at mrchromebox at gmail

nebulakl, 2017-10-17:
how do i get my firmware boot log?

nebulakl, 2017-10-17:
I did some experiments with different SD cards. It seems that booting from a 16G sd card using build-in card reader works properly. However, it failed to recognize 32G and 128G cards. All cards are created by ubuntu's Startup Disk Creator with artful-desktop-amd64.iso.

MattDevo, 2017-10-17:
> how do i get my firmware boot log?  

like I said, send me an email and I'll provide instructions and necessary files - easier to handle that way than on here

closed 2018-09-12 by nebulakl

neilcobbett, 2021-05-15:
I'm having similar issues with my old Toshiba Chromebook2.  I want to boot off the SD card since the internal flash is getting too small for current Linux distros.

I can detect a 32GB SD card but not a 512GB SD card that I would ideally like to use.

I've installed the latest MrChormebox firmware.  The SD card is detected if I unplug it and plug it in again when on the menu page but it wont boot from initial power on.

Is there a tweak I can make to the firmware settings to make it see larger SD cards?

MrChromebox, 2021-05-15:
@neilcobbett doesn't sound like you're talking about SeaBIOS. There's not much I can do to tweak the initial SD card detection in Tianocore without being able to reproduce it here