MrChromebox/firmware #77: OS won't boot if another USB Device is connected

Jimster121, 2017-10-27:
Hello,

I have two USB hubs, one with 4 devices connected and another one with none. When I power up the computer it shows the boot screen and then freezes. When I hit the power button again it turns off. It doesn't have to be a hard drive, any device will do. I have an ASUS C300 Chromebook (Baytrail) with the latest GalliumOS installation.

- Thank you!

ReddestDream, 2017-10-27:
This is not uncommon with some USB hubs and devices plugged in on startup.

Our UEFI firmware, like many other vendor UEFIs, scans connected USB devices for boot devices on power up. If one of the devices is slow or otherwise misses the scan deadline, the firmware will freeze up waiting for the device to respond.

Usually, on our firmware, timeout occurs in less than 2 minutes and the boot process continues. On other firmwares, it can be as long as 5-10 minutes. Depending on the device/timeout/other factors, the firmware may simply freeze waiting on the USB device.

The more USB devices you have plugged in at power on, the greater the odds that this issue will occur. I've found generally that older USB devices and especially USB 2.0 devices are more likely to cause this issue than USB 3.0 and/or newer devices. Also, the wireless transmitters of USB keyboards and mice, especially older ones, can cause this issue. USB 3.0 interference by 2.4 GHz wifi and BT are also known to cause this issue.

All this to say that, while this is a bug, it is a bug that our UEFI firmware has in common with many other vendor firmwares. The simplest solution is to check for the offending device(s), remembering that it may be the hub itself. If you can't find a specific device, try to reduce nearby 2.4 GHz interference. I hope this helps. :)

Jimster121, 2017-10-27:
Thank for your answering quickly, I tested earlier to see if it could actually boot with my devices connected, it booted but very slow. It may actually try to find a device to boot and checks every one until it finds the hard drive.

sshambar, 2022-05-06:
I'm hitting this issue with USB attached HDDs when the drive is in standby (spun-down).  

Device: Asus Chromebox M115U (CN60) running the current release (4.16).  Root drive is internal M.2 SSD.
Exterenal disk: Rosewill RX-358 USB enclosure with a WD Red NAS 12TB drive (data drive)

- USB external disk attached, enclosure is on, and the disk is not spinning
- Chromebox is turned on
- USB disk is triggered to spin-up
- Display shows rabbit, but the "Press ESC..." message does not appear
- Chromebox hangs indefinitely (no timeout - I've left it for over a day...)

If I repeat the process without the USB disk attached (or off), Chromebox boots normally.
If I repeat with the USB drive already spinning, the Chromebox boots normally.

... so the hang appears to be related with the triggering of the USB spin-up.  The problem is when the power fails and then restored, there's a race condition that randomly results in this hang occurring, and so the chromebox sometimes fails to boot after power failures.

The only workaround I've found is actually adding a delay to the power applied to the USB enclosure (via a multi-second delay relay on the plug)... but that's a serious hack :)

Please let me know if I can provide any further debugging details that might help resolve this bug!

MrChromebox, 2022-05-06:
I'm not sure this is something I can resolve in firmware, I have no way of getting debug logging out the device when this happens

sshambar, 2022-05-06:
I don't think the Chromebox has a serial header... I have a USB serial port, but that would require the bios to support that :)

I'll see if I can reproduce with other external USB drives... if so it may be possible to easily reproduce on a device that does have serial headers if the bios can use that for debugging...

MrChromebox, 2022-05-06:
it doesn't, just the pins (but no header connector) for the Google servo debugger

RoboKaren, 2023-07-05:
I'm encountering the same feature/bug as well on Lillipup when a USB-C hub is plugged in (no drives). 

'Wouldn't a good workaround be a UEFI / firmware option that turns off USB port scanning? It'd also be a safety/security feature as well. I wouldn't want my machine booting off a rogue USB drive without me enabling that in firmware.

RoboKaren, 2023-07-05:
Another option would be to decrease the USB timeout as a user-selectable option.

ellyq, 2023-07-06:
First of all, this has nothing to do with this issue (TGL vs BSW).
Second, install release 4.20 and test.

I have confirmed USB4 functionality on ELDRID with Anker 7-in-1 hub.

closed 2026-11-04 by MrChromebox