MrChromebox/firmware #78: Sound problem on ASUS C300 (Baytrail)

Jimster121, 2017-10-31:
Hello, 

I tested **all** the firmwares available, RW_LEGACY, BOOT_STUB and now Full ROM. Since I installed the Full ROM update I had to re-install the OS (Latest version of GalliumOS). When I plug in the headphones, I get distorted sound, this also affects the microphone. **As a workaround**, I unplug the headphones (still getting distorted sound, this time on speakers), put a video or sound file playing, turn up the volume and after a while I get perfect sound. If I connect the headphones again, I still have sound. This usually happens on PlayOnLinux/Wine programes and video/audio calling software (ex. Skype). Rarely happens on Firefox, when I plug the headphones (ex. watch a YouTube video). At the GalliumOS forums people claims that this problem is fixed with the BOOT_STUB or Full ROM update. Why isn't this problem fixed on my device? I already created an issue on GalliumOS forums, but I get no answer. :disappointed:

 - Thanks

MattDevo, 2017-11-01:
this is an OS/driver/config issue, not a firmware one

closed 2017-11-01 by MattDevo

Jimster121, 2017-11-01:
Okay, thanks for clearing that up! 👍