MrChromebox/firmware #79: how to boot from intern harddrive on GNU GRUB version 2.00

rik216, 2017-11-04:
when I set a iso on a usb stick and put in in a chromebook it wont automatically boot the os installation wil start up when i start it up in the boot menu but then i cant start the os / leaving the bios

i have a dell chromebook 11

MattDevo, 2017-11-04:
sounds like you're not preparing the USB properly - you can't simply copy the ISO to the USB, you have to use a program to write it (like Rufus or Etcher)

rik216, 2017-11-05:
hello, Matt 
The problem is when I install the Os on my hard disk it won't boot when I restart the system it still bott's into the bios.

rik216, 2017-11-05:
nevermind im in the os but when i shutdown the os everything is gone when i boot it up again

MattDevo, 2017-11-05:
I think entry #2 is what you're running into, but hard to tell from your description: 
https://mrchromebox.tech/#faq

rik216, 2017-11-05:
no the galiumos is there but all the programs I had installed where gone

MattDevo, 2017-11-06:
> no the galiumos is there but all the programs I had installed where gone

I really don't know what you're trying to say, but it sure doesn't sound like a firmware issue

rik216, 2017-11-10:
i think ive got the problem my chromebook (dell chromebook 11 candy) does only boot from the usb stick

closed 2024-04-28 by MrChromebox