MrChromebox/firmware #80: Chromebox with Legacy now constantly rebooting/resetting

2devnull, 2017-12-06:
Hi,
I installed the legacy firmware, was able to boot a live CD and install LEDE on the internal SSD, however, once LEDE booted, moments later it rebooting and now the system keeps rebooting. Even if I boot from the Live CD again, the system constantly keeps rebooting. What could have gone wrong?

Thanks.

MattDevo, 2017-12-06:
same issue with a different live USB (different distro)?

2devnull, 2017-12-06:
Yes, I tried it with GRML .

2devnull, 2017-12-06:
So, I have it unplugged all night and just booted it into LEDE from the internal SSD (didn't put the liveUSB in). It has not rebooted thus far. If you can, can you shed some light what might happening and why it did what it did last night. I am just hoping this isn't a stability issue as I would like to invest some time in making this work with LEDE.

Thanks for your help.

MattDevo, 2017-12-06:
I've not seen any issues like this that weren't hardware related.  The firmware is rock solid, I have Chromeboxes here that have been running for over 3 years now

2devnull, 2017-12-06:
I know, I have several that have been working for about that time also, so I believe in them, just wasn't sure if something recent with the bios made this unstable.

Assuming it is/was hardware related, what area of HW would/should I investigate?

2devnull, 2017-12-06:
Well, with an uptime of 2hr 52 min, it started to do it again. I had left it, went out and came back and did an 'uptime' command and it started this bool loop again.

MattDevo, 2017-12-06:
I would remove and re-seat the RAM to start, and perhaps try one stick at the time if running two

2devnull, 2017-12-06:
Yes, did that already also. How can I recover to this, to try back stock or UEFI?

2devnull, 2017-12-06:
OK, I tried RAM from a working CBox and the same thing is still happening. So far power adapter and RAM can be ruled out.

MattDevo, 2017-12-06:
in that case, the only safe way to flash it with anything else is using an external programmer.

1.5v RAM won't work (has to be 1.35v DDR3L), and obv it can't boot without RAM.

2devnull, 2017-12-07:
I believe I may have located the issue. I disconnected the power switch cable from the motherboard after it is booted up and it stops rebooting (at least for now). I see that cable goes to a very tiny circuit board which the physical switch is on. Not sure if this is a firmware issue or physical circuit board issue or cable issue. I suspect the latter as it seemed to have be caught in one of the four screws that holds the cover for the case. Not sure where to get a replacement or I can just try to use wake-on-lan and poweroff commands.

MattDevo, 2017-12-07:
that definitely could do it

2devnull, 2017-12-07:
uptime 3:56. Now to locate a parts supplier for this. Those wires are so tiny I am not sure I can repair it.

BTW - Does the CB support WoL?

MattDevo, 2017-12-07:
WoL via ethernet (not wlan), but only from S3 (sleep), not from S5 (soft-off)

2devnull, 2017-12-07:
hmm, so basically if this is first plugged in it will be in S5 mode and therefore no WoL? That will be a problem.

2devnull, 2017-12-07:
Matt, thank you for all your help. 14 hrs without a reboot. Will assume it is the power switch cable for now and close this issue.

closed 2017-12-07 by 2devnull