MrChromebox/firmware #82: HP CB1 (ZAKO) not detecting RAM >2GB when headless

albeauregard, 2017-12-17:
HP Chromebox CB1 (ZAKO)
Upgraded RAM from 2GB to 4GB
Using Full ROM/ UEFI (MrChromebox 07/17/2017)
Linux Mint 18.3 sylvia amd64

When booted with HDMI connected, full 4GB is recognized by the kernel
When booted headless with HDMI disconnected, only 2GB is recognized by the kernel

Notable difference in kernel messages is:

With HDMI connected:
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000178dfffff] usable

With HDMI disconnected:
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x0000000178dfffff] reserved

I'm not sure if this is a firmware issue or a linux kernel issue.  I've tried specifying different memmap= kernel options, but have not had success.

MattDevo, 2017-12-28:
I'll have to see if I can reproduce this on one of my boxes here

albeauregard, 2017-12-28:
Let me know if there’s anything else I can test or provide to help.

marthoc, 2018-01-10:
I think I’m noticing the same issue - Asus Chromebox M004. I’ve upgraded to 8GB ram and when running headless Debian 9 reports only 2GB ram, though lshw reveals both ram modules:
```
        *-bank:0            
             description: SODIMM DDR3 Synchronous 1600 MHz (0.6 ns)
             vendor: Unknown (0)
             physical id: 0
             serial: None
             slot: Channel-0-DIMM-0
             size: 4GiB
             width: 64 bits
             clock: 1600MHz (0.6ns)
        *-bank:1
             description: SODIMM DDR3 Synchronous 1600 MHz (0.6 ns)
             vendor: Unknown (0)
             physical id: 1
             serial: None
             slot: Channel-1-DIMM-0
             size: 4GiB
             width: 64 bits
             clock: 1600MHz (0.6ns)
```

albeauregard, 2018-01-10:
Are you seeing the same sort of messages in dmesg ( dmesg | grep e820 ), where memory above 2GB is being marked as reserved?

MattDevo, 2018-01-10:
at this point, I'm going to say it's a known issue when using the UEFI firmware in headless mode.  My recommendation would be to use the Legacy Boot firmware instead, as that won't have any issues with RAM detection (though it will require an OS reinstall)

marthoc, 2018-01-10:
@albeauregard Yes I am seeing the same reserved messages. @MattDevo thanks for answering my unasked question - I’ll work around this by reflashing to Legacy and reinstalling for now.

albeauregard, 2018-01-18:
Interesting note:

I switch from Mint to Fedora Server (for unrelated reasons) and the issue went away.
While it certainly seems like the firmware should be at fault, there's obviously a connection with the kernel or grub as well.

MattDevo, 2018-01-29:
this issue was identified as a Tianocore bug caused by one of our patches, and will be fixed in the next/upcoming release.  The bug caused all RAM above 2GB to be marked as system/reserved when booted without a connected display.

closed 2018-01-29 by MattDevo