MrChromebox/firmware #83: [Request] Custom splash screens

Leron4551, 2017-12-27:
I'm still Fairly new to GitHub so not sure if this is the place for this, but is there any possibility of adding a way for us to choose our own splash screen for the SeaBios UEFI screen to replace the hare?

MattDevo, 2017-12-27:
the coreboot logo (a European brown hare) is compiled into the Tianocore UEFI firmware payload; to change that, you'd have to recompile the payload/repack the firmware.  If you just want to update the logo while Windows boots, you can use something like HackBGRT to do so, modifying the path/filename to use the one required by the Chromebook UEFI firmware.

closed 2018-12-31 by MrChromebox

GLM1986, 2024-03-22:
Will HackBGRT, available on GitHub work to change MrChromebox UEFI Fireware "Rabbit" Logo, not the Windows 10/ Logo? I have 3 Chromeboxes already flashed with Windows 10 22H2 on one and Windows 11 23H2 on the other two, but would like to change the Rabbit Boot Logo. Also would like to know the Format for the new logo and size if it is possible to use HackBGRT to do this ?