MrChromebox/firmware #84: ASUS Chromebook Flip C302C - No ESC or MicroSD boot

sock5puppet, 2017-12-28:
I've been working on this for a bit and seem to have hit a snag.

I followed instructions here: http://0x3f8.com/2017/02/18/kali-linux-on-the-new-asus-c302ca/

I've run the script, and updated to Legacy_FW (Option 1) as described.

I try to reboot/boot to the MicroSD card (32GB) at which point I see the SeaBIOS splash screen referencing Chromebox and then it goes to "Booting from Hard Disk..." and just stays there. I've tried pressing ESC at various timings after hitting CTRL+L, none of which seem to work.

I never see any message about pressing ESC to select a boot device.

Any thoughts on what I might be doing wrong?

MattDevo, 2017-12-28:
the 'Press ESC for boot menu" only appears when there are multiple boot options available.  Since you're not seeing it, it means that SeaBIOS isn't detecting the SD card (or controller) correctly.  If you're able to boot from USB, then I can instruct you on pulling the firmware boot log so I can see what's causing the problem.

liuxiawei, 2018-01-23:
This is protect screw in c302ca , you should take it out and use "sudo su" and "flashrom --wp-disable" to close the write protection. I had taken it out and run the script then.Now I can use it.

MattDevo, 2018-01-23:
the firmware write protection has nothing to do with the functionality of the RW_LEGACY firmware

namsohj, 2018-06-07:
I have exactly the same problem, do you know any way to solve it so that you can boot from the micro SD and be able to install Linux?

MattDevo, 2018-06-07:
@namsohj the issue is with SeaBIOS' detection/init of some high-speed/high-capacity SD cards, there's nothing I can do about it.  Boot from a USB stick instead, use a A->C adapter if needed

closed 2018-06-07 by MattDevo