MrChromebox/firmware #89: Full brightness backlight behavior during critical low battery charge on lumpy

DrMattChristian, 2018-02-06:
Hi Matt,

Thanks for all the work you do to bring open UEFI/firmware and booting other OSes to Chromebooks!

I have an old Samsung Chromebook Series 5 550, AKA [lumpy board](https://www.chromium.org/chromium-os/developer-information-for-chrome-os-devices/samsung-sandy-bridge), running [GalliumOS 2.1](https://wiki.galliumos.org/News/GalliumOS_2.1) with your firmware 2017.07.14.

Due to hardware age, when on battery power, the actual capacity is about half of full Wh design. This means that a low battery can be a common occurrence on this model.

When the battery charge reaches a critical "red" low power stage (few % left) and power manager calls for suspend/hibernation, the backlight is changed to **full brightness for several tens of seconds**. When this happens, the **brightness keys are also unresponsive** so I cannot dim manually.

After another number of seconds, the display shuts down (goes black) and suspends/hibernates as expected. This super-bright backlight only happens during automatic suspend handling of critical low battery, not when charged or plugged in.

Let me know what information or logs you need to better understand what's going on. If you have lumpy hardware laying around then perhaps you can try to reproduce this behavior as well.

I just read that you just released a new firmware 2018.02.04 but I haven't tried it yet. I presume most of the major fixes/updates are to help much newer hardware and add board support. However the other UEFI/OpenBoot updates might help older models as well.

Sincerely,
(Another) Matt

MattDevo, 2018-02-07:
I don't see anything here that's due to firmware/firmware settings; I'd re-post this on /r/GalliumOS

closed 2018-02-07 by MattDevo

DrMattChristian, 2018-02-07:
Okay understood. I updated to your UEFI firmware 2018.02.04 anyway.

I'm also updating GalliumOS via apt in case it's a software package issue, hundreds of debs need to be updated. I read that newer Linux kernel in Gallium OS release repo is just around the corner.

I will attempt to reproduce the behavior at least one more time, after all updates are installed, before re-posting to Gallium sub-Reddit.